UltimateHackingKeyboard/firmware #2: Write/read configuration via USB

mondalaci, 2016-10-22:
The configuration has to be transferred between Agent and the UHK.

The writing of the configuration will happen as follows:
1. Agent sends the configuration to the UHK over USB in a series of 64 byte packets. The firmware saves these chunks in the RAM of the UHK.
2. Agent sends a "validate configuration" command to the UHK. The UHK validates the configuration by executing the configuration parser in dry run mode.
3. If the configuration parser succeeded then Agent sends a "save configuration" command to the UHK which will persist the validated binary configuration from its RAM to the EEPROM.

The reading of the configuration will happen as follows:
1. Agent reads the configuration from the UHK over USB in a series of 64 byte packets.
2. Agent validates the binary configuration and proceeds further if it's valid.

I'll implement the firmware part of the above process and write up some crude JavaScript scripts to demonstrate its use. I'll implement the "validate configuration" part as a dummy command because the configuration parser will only be available later.

jmrtns, 2017-01-24:
It would be great to implement a means for users to sign the firmware to prevent unwanted and potentially malicious firmware flashes when plugging the keyboard into unknown hosts. This would also require a means of doing a hardware based reflash of the public key stored on the keyboard or of the entire firmware as it would be pretty bad to loose your private key and have a bricked keyboard. I don't see a means of hardware re-flash as a security bug because if someone malicious has that much physical access to your keyboard then it's no longer your keyboard anyways ;)

A plus side to this besides added security is the likely ability to market it's security as "The first keyboard to have user signed firmware" (assuming this is true which, to my knowledge, is).

algernon, 2017-01-24:
The problem is that doing crypto in firmware is... non-trivial. Rolling a key checker from scratch would be a terrible idea, so this'd need an existing library, such as [NaCl](http://nacl.cr.yp.to/features.html). But it still needs to fit into the firmware, and a crypto library, however small, will still require a noticable amount of space.

It would be neat, nevertheless, and such a step would be reasonably easy to introduce too - the hard parts are the crypto bits.

mondalaci, 2017-01-24:
We'll implement multiple security modes according to https://ultimatehackingkeyboard.com/blog/2015/02/26/securing-firmware-upgrades . The specification is not final but it should give you a good idea.

I think the above security modes should be more than sufficient, and signed firmware images would be very hard to get right, so it wouldn't really worth having them.

@jmrtns Please open a new issue next time, unless you find a relevant one which you're welcome to comment. Firmware signing is not relevant to this issue.

mondalaci, 2017-08-11:
This has been implemented.

closed 2017-08-11 by mondalaci