UltimateHackingKeyboard/firmware #15: Handle layer switcher keys the right way

mondalaci, 2016-12-11:
There are 4 layers per keymap. The base layer is active by default when no layer switcher key is pressed. There are also the Mod, Mouse, and Fn layers activated by the Mod, Mouse, and Fn keys respectively.

When multiple layer switcher keys are pressed the first one dominates. I mean the one that gets pressed before the others. This allows us to switch to the Mouse layer by holding the Mouse key pressed, then also hold the Mod key to accelerate the movement of the pointer according to [the factory keymap](https://ultimatehackingkeyboard.com/#layers).

algernon, 2016-12-11:
Hrm. So if I press and hold `Mod` then `Mouse`, then `Mod` would be active? That's pretty easy to accomplish: we introduce an *"It's dead, Jim"* key, which has no function, and if seen, the lookup function won't look it up on the base layer, either. This way we can disable the other layer keys on each layer. I'd call it `Skip`. Could even have its own type for now, to make things easy.

Turning them into other keys in another layer already works (or well, will work, when #16 is merged).

So, in short, this is easily doable for the factory keymap. I would not, however, restrict layer behaviour to the one described, because one may want another behaviour: for example, if I hold `Mod`, I may still want `Mouse` to work. Not exactly sure why, but still.

I'll do this ~tomorrow, most likely.

mondalaci, 2016-12-12:
I've tried to follow your description but couldn't understand. Please let me propose what I think is the simplest way to make this happen.

Let's store the active layer in `uint8_t activeLayer`. According to `layer.h` its value would be `LAYER_ID_BASE` first. Then we'd iterate over the key states over and over again, look for layer switcher keys and build up a `uint8_t pressedLayerSwitcherKeys[LAYER_COUNT-1]` array every time. If `activeLayer` is already pressed according to `pressedLayerSwitcherKeys` then it shouldn't be changed, otherwise it should according to the pressed keys.

I was thinking a lot about other ways to make this happen, including enabling the user to customize the prioritization of layer switcher keys, but alternative implementations would introduce further user options for no good reason. I'm fairly confident that this policy should suit the needs of almost every user.

But before anyhting else, it'd be worth to implement #17 to minimize further refactorings.

algernon, 2016-12-12:
Mm, no, I was thinking of a different way. I wouldn't enforce the layer ordering on the firmware level. I'd enforce it by the various layers not having keys that switch to the others. So if you have `Mod` pressed, there are no `Fn` or `Mouse` keys. That way, there is no problem with ordering, because you only ever have one layer active.

Less code, and easier to understand how layers interact (they simply do not).

algernon, 2016-12-12:
On the way to work, this issue kept coming back, and I think I have something between our two ideas, which should work well for momentary layer switchers, but falls a bit short on toggles, I'll explain in a bit why.

So, the idea is that we don't track the state of all layers. We just have an `activeLayer`. If a layer switcher is pressed when `activeLayer` is not `LAYER_ID_BASE`, it would simply do nothing. We'd also remove layer switcher keys from the layers themselves, so this would never happen with the factory firmware. But if someone remaps their layout, it could, and if I'd press `Mod` then `Mouse`, `Mod` would be active. But if I keep holding `Mouse`, and release `Mod`, then first `Mod` turns off, and `activeLayer` becomes zero, and since `Mouse` is held, it will now become the active layer. This works for any number of momentary layer switcher keys.

For toggles, the case is a bit different, because they are not held, so can't re-attempt the switch. But there may be ways around that... like your `pressedLayerSwitcherKeys` idea. I'll have to think/experiment a bit with that.

mondalaci, 2016-12-12:
I like your thinking a lot, and would go as far as disallowing users to assign layer switcher keys on non-base layers in Agent. It'd solve a lot of problems.

We definitely have to address what should happen when a layer switcher key gets released and there are still other keys pressed. Just imagine pressing Mod+A, and accidentally releasing Mod first. We don't want "A" to be sent to the host.

A while back, [I implemented this](https://github.com/UltimateHackingKeyboard/firmware-avr/blob/master/right/KeyboardReports.c#L111) by using a suppress flag per key. The idea is that when the active layer switcher key gets released, all the other keys that are pressed get flagged with a suppress flag. Then the suppressed keys are treated as released keys until they're actually released at which point the suppress flag gets removed.

For this to be implemented, we have to know the previous state of keys, so they must be stored. This makes us store the previous state, the actual state, and the suppress flag per key. How about creating a `key_state_t`, containing these values as bit fields? This would also affect `key_matrix_t` and related functions.

Let's rather not deal with the toggle layer situation right now, and address it in the next step because this is already quite complex as it is.

algernon, 2016-12-12:
I'd deal with suppression separately. The keymap refactor and this are big enough changes on their own.

mondalaci, 2016-12-12:
I agree, good thinking.

mondalaci, 2016-12-12:
Cool, thanks! I reviewed your PR and made some comments about the changes to be done.

mondalaci, 2016-12-13:
> I would go as far as disallowing users to assign layer switcher keys on non-base layers in Agent.

@algernon Do you agree with this? If so, I'll create a relevant issue in the Agent repo.

algernon, 2016-12-13:
To support the behaviour discussed, that would be best, yes.

mondalaci, 2016-12-13:
Amen! Just created https://github.com/UltimateHackingKeyboard/agent/issues/216

Jozsi or Nejc will make it happen.

PR #21 (2016-12-13): Layer handling simplified, and streamlined
This simplifies the layer handling by only allowing a single layer to be active at any one time. As such, the behaviour after this patch should be as described, and as discussed in #15, as far as I understand.

Also moved the layer handling to a separate file, let me know if that is not desired.

Fixes #15, if all goes well.
closed 2016-12-13 by mondalaci