UltimateHackingKeyboard/firmware #27: Disable the NMI pin

mondalaci, 2016-12-24:
According to Jan:

> the default function is NMI_b. You will need to disable NMI in the flashed image, otherwise the chip won't boot sometimes. Found out the hard way. From my notes: "NMI functionality is active by default. To disable it, I had to set the FOPT byte in the flash configuration section of the binary. This byte is then loaded on reset and the system is configured accordingly. Flash configuration can be found in the startup_SYSTEMNAME.S file. The previous-to-last byte is the FOPT register, and setting it to 0xFB instead of 0xFF (0xFF-0x04, to disable bit 2) disables NMI.
> Layout of the flash configurations is described in "Flash Configuration Field Description" in the FTFA section of the RM."

This should be done for the firmware of every module.

Also gotta check whether KBOOT is affected.

mondalaci, 2016-12-26:
Also note that:

>    Yes, if you disable the NMI function, you don't need an external pull-up resistor.
>   Please note, disable the NMI function should modify the flash configuration field,FOPT[2]=0,  the flash address is 0x40d.

According to https://community.nxp.com/message/863431?commentID=863431&et=watches.email.thread#comment-863431

mondalaci, 2017-10-25:
Done by https://github.com/UltimateHackingKeyboard/bootloader/commit/3fcbe24143b37029ae93ed5115e9ca1c2aef62dd

Lauszus, 2018-01-16:
@mondalaci I believe the you have used the wrong value in: https://github.com/UltimateHackingKeyboard/bootloader/commit/3fcbe24143b37029ae93ed5115e9ca1c2aef62dd. 

I believe the right value would be:
```C++
0xFFFFF995
```
As you want to clear the 1 and 2 bit. Please see the datasheet: https://www.nxp.com/docs/en/reference-manual/K22P64M50SF4RM.pdf at page 160.

```0xFC``` clears the ```LPBOOT``` and ```EZPORT_DIS``` bits instead.

mondalaci, 2018-01-17:
@Lauszus Thanks, great catch!

Actually, I think you're not referring the most relevant datasheet which is [this one](https://www.nxp.com/docs/en/reference-manual/K22P121M120SF7RM.pdf), page 178 and 179. There isn't much difference though. The extra FAST_INIT bit shouldn't be enabled, I believe.

I'm not sure about LPBOOT. What's your take on it?

Lauszus, 2018-01-18:
Thank you. I am working on another project using a Kinetis device and your code helped a lot for my understanding how to enable flash protection!

No I do not think you should clear ```LPBOOT``` since your device is not a low power device.

It makes sense that you clear ```NMI_DIS``` if you haven't got an external pull-up on your PCB.

Clearing ```EZPORT_DIS``` should be fine as well, as it does not look like you are using EzPort.

mondalaci, 2018-01-22:
Thanks again so much!

Done by https://github.com/UltimateHackingKeyboard/bootloader/commit/8a67901ae965af3c4a5fd72475e074fad46e8dae

closed 2018-01-22 by mondalaci

Lauszus, 2018-01-22:
@mondalaci you're welcome :)