UltimateHackingKeyboard/firmware #30: Signed firmware

jmrtns, 2017-01-27:
It would be great to implement a means for users to sign the firmware to prevent unwanted and potentially malicious firmware flashes when plugging the keyboard into unknown hosts. This would also require a means of doing a hardware based reflash of the public key stored on the keyboard or of the entire firmware as it would be pretty bad to loose your private key and have a bricked keyboard. I don't see a means of hardware re-flash as a security bug because if someone malicious has that much physical access to your keyboard then it's no longer your keyboard anyways ;)

While this is mostly handled by "multiple security modes according to https://ultimatehackingkeyboard.com/blog/2015/02/26/securing-firmware-upgrades" 

I can still see benefits to the signed firmware model, mainly mass production and the fact that unless you use the same password for the security mode in the keyboard as you do for a common key (maybe not the most secure), It seems likely to forget the password.

Probably as a side note this issue can probably also benefit the libreboot project:
https://libreboot.org/

I'll contact libreboot to see if they have any information on the matter but it may become an issue of me or someone creating a seperate repository, or adapting an existing one, for a firmware module that both and other projects can share. I know there are proprietary implementations of signed firmware on small firmware on devices such as kanguru USB drives so it should be possible but it will probably take quite a bit of time.

At this time though, this issue should probably be pretty low priority since the above security modes handle the issue pretty well and make signed firmware more of a "nice to have" than anything.

mondalaci, 2017-02-22:
I am closing this issue because in all honesty, I don't believe we'll ever have the resources to implement it, nobody else has requested it, and as it was correctly stated above, the security modes will provide a sufficient alternative.

@jmrtns Please feel free to resurrect this issue if actual, solid work gets done on it. I can also vaguely recall that we started a conversation about this in another issue/PR of possibly another repo. You're also welcome to reference that here, too.

closed 2017-02-22 by mondalaci

jmrtns, 2017-03-07:
Sounds good, thanks for the time =)