UltimateHackingKeyboard/firmware #31: Macro ideas

algernon, 2017-01-30:
I've been pondering about macros recently, and the best way to implement them. With other firmware I worked on, they were pretty darn simple, because they are compiled into the firmware itself. With the UHK, they're not, as they can come through the Agent. This presents some non-trivial problems:

- We can't reasonably pre-allocate memory for the macros, because that'd impose serious limits on what one can do. Not to mention that the limit may well be off, by a lot, and in a lot of cases, be terribly wasteful.
- We do not want to dynamically allocate memory, even if we could, because it would become very hard to predict how much memory we actually have to allocate from.

So, I thought: what if we didn't copy macros from EEPROM to RAM, like we do for the keymaps? What if we re-played macros from EEPROM directly? We'd still need a lookup table in RAM, that tells us which macro is where, so we would have to parse only a fraction of the config (or we could have a table of contents in EEPROM too, supplied by the Agent, to make this even easier on the firmware), but the macro would be replayed from EEPROM. While EEPROM access may be slower than RAM, in case of macros, I do not think that matters much - delays will have to be introduced anyway, otherwise playback would be faster than what the host is ready to handle.

So my proposal is that I'll extend the config parser to also parse macros, and store their location in a pre-allocated array (allowing up to, say, 64 macros). Then implement a macro-playback method on top of this.

What do you think?

mondalaci, 2017-01-30:
Your proposal is totally in line with the way I'd have implemented this. :)

Luckily, you won't even have to touch the EEPROM on a regular basis because the plan is to copy the content of the EEPROM into the RAM right upon firmware startup. Then you'll be able to deal with the RAM directly.

On a related note, according to my evil plans, the content of the EEPROM will be stored twice in the RAM. One will contain the actual configuration, and the other will contain a temporary configuration that gets transferred by Agent in small chunks. Upon saving the configuration, Agent will validate the temporary configuration, then save it into an EEPROM as quickly as possible in about 0.1s. This is to minimize the possibility of data corruption. Admittedly, these arrays will occupy half of the RAM (2x32KB of 128KB) but we can afford it, and it makes implementation easier, faster, and safer.

Let's rather allow 256 macros instead of 64. There's still plenty of RAM left.

As for the next step, given that Agent is already able to transfer both the actual layer and the actual keymap, maybe it would be worth the make the firmware parse keymaps. Then you could go further upwards in the configuration tree and parse everything, macros included.

Thanks a ton!

mondalaci, 2017-01-30:
I forgot to comment on a suggestion of yours:

> We'd still need a lookup table in RAM, that tells us which macro is where, so we would have to parse only a fraction of the config (or we could have a table of contents in EEPROM too, supplied by the Agent, to make this even easier on the firmware)

As a design principle, we'd rather avoid making Agent generate any redundant information in order to utilize the EEPROM space as efficiently as possible, so it's left to parser to extract the macro offsets.

algernon, 2017-01-30:
I figured that part from your 256 vs 64 macros comment =)

mondalaci, 2017-07-21:
The macro paser has been implemented in the meantime, so I'm closing this issue.

closed 2017-07-21 by mondalaci