UltimateHackingKeyboard/firmware #38: CMakeLists not updated

stephengroat, 2017-05-04:
the contents of `left/build/armgcc/CMakeLists.txt` seem to be out of date.

could those be updated? i'm trying to add travis, but i need to use the `armgcc` build system instead of the `kds` system

mondalaci, 2017-05-04:
Hi Stephen,

Admittedly, both `{left,right}/build/armgcc/CMakeLists.txt` are terribly out of date. I'd love to update them eventually, but right now, there's a lot on my table, and I can't promise you a definite ETA.

In case you want to wrap this up, installing KDS, building the firmware in it, and reading the build log may be quite helpful. Should you need any further help, I'm always here for you.

Thanks!

stephengroat, 2017-05-05:
i can't get the kds registration process to work.

could you post a build from clean log for me :pray:

mondalaci, 2017-05-05:
Sure thing! [Here you go.](https://gist.github.com/mondalaci/a6af345283c7d432b404c0e652cf5709)

stephengroat, 2017-05-05:
any advice on [this](https://travis-ci.org/stephengroat/firmware)? some sort of linker issue, not 100% sure

mondalaci, 2017-05-05:
Looks like you've already made the left firmware build in Travis. Great work!

As for the rest, I'm quite overwhelmed by this huge log. Resolving endian related warnings would be very useful.

But I assume the following is the relevant error message:

/home/travis/build/stephengroat/firmware/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: section m_usb_bdt VMA [000000001fffa200,000000001fffa3ff] overlaps section .bss VMA [000000001fff1e10,000000001fffaa03]

Looks like this is a linker related issue. Some memory regions seem to overlap. Unfortunately, I have never encountered this issue. Maybe it'd be worth to grep for the memory section names and dig deeper.

Alternatively, it'd be also worth to restore a much earlier revision of the firmware repo because a while back the armgcc build did work for both the left and right firmwares which might have been about a year ago. I was fiddling with the files of the armgcc directory, and chances are I haven't touched those files ever since - that's how far you should go back. This way, you could have a working base and move forward from there.

Hope it helps!

stephengroat, 2017-05-08:
found a way to build eclipse from the command line

closed 2017-05-08 by stephengroat