UltimateHackingKeyboard/firmware #44: Fix BusPal

mondalaci, 2017-06-24:
Currently, BusPal partially works. For example:

```
$ blhost --usb 0x1d50,0x6121 --buspal i2c -d get-property 1
Inject pre-config command 'i2c'
[01 00 0c 00 c1 00 00 02 10 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<03 00 0c 00 a0 00 00 02 00 00 00 00 c1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
Successful generic response to command 'i2c'
  - took 0.000 seconds
Response status = 0 (0x0) Success.
Inject command 'get-property'
[01 00 0c 00 07 00 00 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<03 00 0c 00 a7 00 00 02 00 00 00 00 00 00 01 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
Successful response to command 'get-property(current-version)'
  - took 0.000 seconds
Response status = 0 (0x0) Success.
Response word 1 = 1258356736 (0x4b010000)
Current Version = K1.0.0
```

Unfortunately, the `flash-image` command aborts:

```
$ blhost --usb 0x1d50,0x6121 --buspal i2c -d flash-image uhk-left.srec 
Inject pre-config command 'i2c'
[01 00 0c 00 c1 00 00 02 10 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<03 00 0c 00 a0 00 00 02 00 00 00 00 c1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
Successful generic response to command 'i2c'
  - took 0.000 seconds
Response status = 0 (0x0) Success.
Inject command 'flash-image'
Wrote 192 bytes to address 0
[01 00 0c 00 04 01 00 02 00 00 00 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<03 00 0c 00 a0 00 0c 02 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
Successful generic response to command 'write-memory'
[02 00 20 00 00 06 00 20 e9 04 00 00 45 05 00 00 49 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
(1/3)16%[02 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 4d 05 00 00 00 00 00 00 00 00 00 00 51 05 00 00 55 05 00 00]
(1/3)33%[02 00 20 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00]
(1/3)50%[02 00 20 00 59 05 00 00 41 05 00 00 5d 05 00 00 41 05 00 00 61 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00]
(1/3)66%[02 00 20 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00]
(1/3)83%[02 00 20 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00 41 05 00 00]
(1/3)100% Completed!
<03 00 0c 00 a0 00 00 02 01 00 00 00 04 00 00 00 49 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
  - took 0.001 seconds
Response status = 1 (0x1) Failure.
```

Strictly speaking, BusPal may be flawless, but [errata ID 806](http://www.nxp.com/assets/documents/data/en/errata/KL03Z_1N86K.pdf) is probably the culprit, so it should be worked around by BusPal. You can read more about the erratic behavior in a [related NXP forum post](https://community.nxp.com/thread/450153).

I've created the the `buspal-fix` branch to easily reproduce the above issue. [My commit](https://github.com/UltimateHackingKeyboard/firmware/commit/928b86db8e1349eaf30f26ec60e06cd42162c095) merely sets the timeout of the left bootloader to the maximum 65 seconds and enumerates the right keyboard half as the BusPal proxy over USB.

ErichStyger, 2017-07-19:
`flash-image` seems not to be supported by the KL03Z ROM bootloader:
```
blhost --usb 0x1d50,0x6121 --buspal i2c -d get-property 7
Inject pre-config command 'i2c'
[01 00 0c 00 c1 00 00 02 10 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<03 00 0c 00 a0 00 00 02 00 00 00 00 c1 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
Successful generic response to command 'i2c'
  - took 0.006 seconds
Response status = 0 (0x0) Success.
Inject command 'get-property'
[01 00 0c 00 07 00 00 02 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00]
<03 00 0c 00 a7 00 00 02 00 00 00 00 6b 1d 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00>
Successful response to command 'get-property(available-commands)'
  - took 0.011 seconds
Response status = 0 (0x0) Success.
Response word 1 = 7531 (0x1d6b)
Available Commands = flash-erase-all, flash-erase-region, write-memory, flash-security-disable, get-property, execute, reset, set-property, flash-erase-all-unsecure
```

What I have been working on is to get the `write-memory` command working. It is the same as `flash-image` but requires binary files (see [Binary (and S19) Files for the mbed Bootloader with Eclipse and GNU ARM Eclipse Plugins](https://mcuoneclipse.com/2014/04/20/binary-files-for-the-mbed-bootloader-with-eclipse-and-gnu-arm-eclipse-plugins/) so I hope this is not an issue?

ErichStyger, 2017-07-19:
I have commited a fix for this issue:
https://github.com/UltimateHackingKeyboard/firmware/commit/ccc02fe795b128228fa77285fdc66f3e86655f7c

It works now fine on my side:
```
----------------------------------------------------------------
Process to update the KL03Z (left keyboard):
Bring KL03Z into bootloader mode: plug in the USB cable and within the first 3 seconds issue the following command:
blhost --usb 0x1d50,0x6121 --buspal i2c get-property 1

Inject pre-config command 'i2c'
Successful generic response to command 'i2c'
Response status = 0 (0x0) Success.
Inject command 'get-property'
Response status = 0 (0x0) Success.
Response word 1 = 1258356736 (0x4b010000)
Current Version = K1.0.0
----------------------------------------------------------------
Erase and unsecure flash:
blhost --usb 0x1d50,0x6121 --buspal i2c -d flash-erase-all-unsecure

Inject pre-config command 'i2c'
Successful generic response to command 'i2c'
Response status = 0 (0x0) Success.
Inject command 'flash-erase-all-unsecure'
Successful generic response to command 'flash-erase-all-unsecure'
Response status = 0 (0x0) Success.
----------------------------------------------------------------
Program new binary:
blhost --usb 0x1d50,0x6121 --buspal i2c write-memory 0x0 uhk-left.bin

Inject pre-config command 'i2c'
Successful generic response to command 'i2c'
Response status = 0 (0x0) Success.
Inject command 'write-memory'
Preparing to send 15792 (0x3db0) bytes to the target.
Successful generic response to command 'write-memory'
(1/1)100% Completed!
Successful generic response to command 'write-memory'
Response status = 0 (0x0) Success.
Wrote 15792 of 15792 bytes.
----------------------------------------------------------------
Reset KL03Z:
blhost --usb 0x1d50,0x6121 --buspal i2c reset

Inject pre-config command 'i2c'
Successful generic response to command 'i2c'
Response status = 0 (0x0) Success.
Inject command 'reset'
Successful generic response to command 'reset'
Response status = 0 (0x0) Success.
----------------------------------------------------------------

```

mondalaci, 2017-07-19:
Thank you so much, Erich! Your fix seems to be working perfectly. This is a huge step forward!

closed 2017-07-19 by mondalaci