UltimateHackingKeyboard/firmware #70: 6KRO + NKRO USB interfaces

mondalaci, 2017-12-10:
Currently the firmware exposes a basic 6KRO keyboard USB interface. We'll expose an NKRO keyboard interface, but its compatibility will be limited (no BIOS support) so it'd probably be better to to leave the 6KRO keyboard USB interface instead of replacing it, and add an NKRO interface, too. Then the firmware would detect which interface is polled. If the NKRO interface is polled, then use that. Otherwise fall back to the 6KRO interface.

iprok, 2018-06-23:
I have MSI GS70 laptop. My BIOS is password protected and I can't enter password for it. When I press a button 11-13 symbols goes to input field. Is it related to this issue or no?

eltang, 2018-06-23:
I believe that would be #87.

mondalaci, 2018-06-23:
@iprok The root cause of the issue should be #72 which has been fixed but at the same time #122 emerged. We can't release a new firmware version until the latter gets fixed.

lonetech, 2018-08-19:
There should be no need to "detect" which interface is polled. Either set_configuration or set_protocol should simply select the appropriate one (BIOSes using boot protocol are required to send both to select the boot protocol reports). Ideally you'd just present a 64 byte interrupt endpoint (permitted in full speed, not low speed) and your firmware could have the only difference in the report formats be that 62-key rollover uses the full size array. Another option is to go full bitmap, in which case you could cover far more keys in a smaller report. I'm not familiar with a keyboard protocol requiring more than about 20 keys simultaneously, but N>6.

mondalaci, 2018-08-20:
@lonetech Thanks for the idea! I'm not as familiar with the USB protocol as you are, but I think I mostly understand your suggestion. The thing I don't understand is the "full bitmap" part. Is this a way to represent a key state with a bit in the report? If so, I'd assume it's far less widely supported by BIOSes / OSes than the regular way in which a byte describes a scancode. Would you please comment on this?

lonetech, 2018-09-02:
BIOSes support boot protocol specifically, but within that there's both a bitmap section for shift keys and a key code array section for other keys. You can find the format specification in appendix B.1 of the [HID specification](http://www.usb.org/developers/hidpage/HID1_11.pdf). The bitmapped range is key codes 224-231, which we find at the end of the HID usage tables for keyboard (page 59 of [HID usate tables version 1.12](http://www.usb.org/developers/hidpage/Hut1_12v2.pdf)) to cover Left Control/Shift/Alt/GUI then Right in the same order. This is why those keys don't count for the N in NKRO, because they're special cases. The HID report descriptor format would allow you to make a bit array for 512 keys or so, which is over twice the defined keys (though some others exist in e.g. Media rather than Application/Keyboard). Operating systems that parse the descriptors rather than use Boot Protocol will support bitmapped form because it's already in use for both keyboards (shift keys) and all sorts of buttons (game pads, telephone handsets, etc).

steamraven, 2021-03-12:
I've been looking through the linux kernel source so see how it would handle large nkey buffers and the bitfield option above.  You can find the source here https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-core.c#L1533

If we wanted full Nkey with scancodes, that would need the full 64 byte usb buffer which is 62 keys (34 right + 31 left + 3 key module - 6 that are sent by modifier = 62). However the linux driver code always runs through the entire buffer and worse, for each byte of the buffer, it runs the length of the buffer twice to see if things changed.  That is 62*(2*62) = 7688 memory access every USB reports.

In contrast the bitfield is processed a single time with direct access to check if the previous state of the key.  However, It does allocate a full 32bits per bit of the bitfield..
 
If we go the bitfield way, we could do 13 bytes which would cover all a standard keyboard. We could leave the 6 key scancode buffer for any special keys in the high scancodes.

I was concerned about key repetition using the bitfield, but I found that is detected and removed in https://elixir.bootlin.com/linux/latest/source/drivers/hid/hid-input.c#L1393

I am a little concerned about the compatibility of using a large bitfield, especially in memory constrained hosts.  It might be good to be able to turn it off with a configuration setting.

I have put together sample code, however, since I am waiting on my UHKv2, I have not tested it yet. 
steamraven@bf614d310f4cedac061ea988ba21aca760c1f4c4
https://github.com/steamraven/firmware/tree/nkey-bitfield

The code implements both the Boot keyboard (which is 6-key rollover) and the normal keyboard.  It switches to Boot when it receives the SetProtocol 1 event (which BIOS should send), switches back with setProtocol 0

mondalaci, 2021-03-13:
@steamraven Thanks for elaborating on your findings! I'd much rather max out the number of simultaneously usable keys and stick to 1 byte per key.

steamraven, 2021-03-13:
Once I fully test it out, I will put in a pull request for other people to test.

mondalaci, 2021-10-13:
@coxley The mentioned problems shouldn't be related to this issue. Please shoot an email to support@UltimateHackingKeyboard.com, and we'll help you.

mondalaci, 2022-02-18:
Resolved by #448.

closed 2022-02-18 by mondalaci

gustavospiess, 2022-04-13:
Please, check that the [knowledge base](https://ultimatehackingkeyboard.com/knowledgebase) wasn't updated and sill links this issue.

https://ultimatehackingkeyboard.com/knowledgebase/does-the-uhk-support-n-key-rollover

mondalaci, 2022-04-13:
We have yet to release a new firmware version that implements NKRO, after which I'll update the knowledgebase article.

spuder, 2023-10-26:
I very much would like to have NKRO. 
Searching the changelog I don't see a release yet (perhaps I'm missing it)

https://github.com/UltimateHackingKeyboard/agent/releases?q=nkro&expanded=true

kareltucek, 2023-10-26:
Well, nkro is a *firmware* feature. You will not find it in *agent* changelog...

mondalaci, 2023-10-26:
See it for yourself that it works: https://www.mechanical-keyboard.org/key-rollover-test/