UltimateHackingKeyboard/firmware #71: Basic + advanced mouse USB interfaces

mondalaci, 2017-12-10:
The current mouse interface is fairly advanced featuring horizontal + vertical scrolling. The problem is that it doesn't seem to be handled by BIOSes. Let's also expose a basic BIOS compatible mouse and fall back to it if needed the same way as for #70.

Lauszus, 2018-02-13:
Adding a hidboot mouse interface should solve this issue.

mondalaci, 2018-09-25:
I've just checked the UHK mouse in the UEFI BIOS of my PC, and it works. This doesn't mean that it works with all BIOSes, but I think adding an extra USB interface just for this niche use case is overkill. One should not spend a lot of time in old BIOSes, and they support keyboard input anyways.

closed 2018-09-25 by mondalaci