UltimateHackingKeyboard/firmware #72: Media key scancode repetition

mondalaci, 2017-12-21:
On Windows {7,8,10} consumer codes are being repeated (quickly toggled) when relevant keys are pressed. I encountered similar issues earlier and though I fixed them all, but apparently not. I'll look into how Windows sends HID reports, and adjust `UpdateUsbReports()` accordingly.

Lauszus, 2018-02-13:
The system keys are sent out continuously as well. Usually a keyboard only sends out a message whenever the report changes.

I believe this is the cause of #85 and possibly also #84.

Lauszus, 2018-04-10:
I believe https://github.com/UltimateHackingKeyboard/firmware/pull/104/commits/15d09552a4d475032869594cf3a0f6eb2674a822 fixes this.

mondalaci, 2018-04-10:
Unfortunately, this issue still stands.

eltang, 2018-04-14:
I'm no USB expert, but I notice that the USB stack sends reports in response to kUSB_DeviceHidEventSendResponse. I was expecting it to rely on kUSB_DeviceHidEventGetReport, based on what I've seen other USB stacks do. What was the rationale for using the current method?

piratk, 2018-05-08:
It seems that this issue also affects launching the calculator, yielding a fair bunch of calculators if used on Windows.

mondalaci, 2018-05-08:
Yes, we know. Calculator is also a media scancode.

Lauszus, 2018-06-06:
@eltang this is how all the other interfaces does it as well and they seem to not have this issue?

Lauszus, 2018-06-06:
I just captured the packets while the USB keyboard was just plugged in. I then held down the volumen up button and then let go a couple of seconds later.

The scan codes are NOT toggled. Everything looks alright.

Are you sure that this is not the expected behaviour on Windows? I know that in older version of Windows you could hold SUPER+e down and you could crash Windows as it would just keep opening a file explorer window. However this is fixed in Windows 10.

Here is the Wireshark capture: [volume_up.pcapng.zip](https://github.com/UltimateHackingKeyboard/firmware/files/2077997/volume_up.pcapng.zip)

Lauszus, 2018-06-06:
Here is the output after applying a filer and only outputting the data: 
[data.txt](https://github.com/UltimateHackingKeyboard/firmware/files/2078085/data.txt)

FYI I exported the data using the following command:

```bash
tshark -r volume_up.pcapng -Y usb.darwin.endpoint_address==0x84 -T fields -e usb.capdata > data.txt
```

Note that you should replace ```usb.darwin.endpoint_address``` with ```usb.bEndpointAddress``` if you are running anything except for Mac.

closed 2018-06-07 by mondalaci

Jos512, 2018-10-12:
PS: This issue is linked from the UltimateHackingKeyboard start guide, saying:

> Windows users, the Media key scancode repetition firmware bug makes media keys repeat very quickly on Windows. We're working on fixing this bug, so please bear with us.

If I understand the above merges correctly, the bug is fixed. So that information can be removed from the start guide. :slightly_smiling_face:

jlsjonas, 2018-11-14:
Why is this issue closed? the merge that fixes it got reverted and the issue still persists for windows users. https://github.com/UltimateHackingKeyboard/firmware/pull/129

Not just for volume, but also for play/pause etc.

PR #120 (2018-06-07): The UHK now only sends out data when it changes
Fixes #72, fixes #84 and fixes #85
commit 10985abc (2018-06-07): The UHK now only sends out data when it changes

Fixes #72, fixes #84 and fixes #85