UltimateHackingKeyboard/firmware #73: Keyboard interface not working when USB enumerated while Agent is running on Win 7

mondalaci, 2017-12-22:


luteijn, 2018-08-02:
I can confirm I've seen this happening on  my Win7 laptop at work. Not a big issue for now as I don't usually have/leave agent running. I expect this will get fixed eventually as the os will be upgraded 'soon'.

mondalaci, 2018-09-25:
I originally opened this issue because I thought we should deal with every bug, no matter what, but upon further consideration, I'm confident that we should close this.

According to the [Windows lifecycle fact sheet](https://support.microsoft.com/en-us/help/13853/windows-lifecycle-fact-sheet), Windows 7 mainstream support ended on January 13, 2015 and extended support will end on January 14, 2020.

Given these dates and our constrained resources, I consider spending time on this issue a waste of time, so I'm closing it.

closed 2018-09-25 by mondalaci