UltimateHackingKeyboard/firmware #74: Left keyboard half stops working

mondalaci, 2017-12-23:
**If you're reading this and you're affected, please subscribe to this issue! A new firmware will be released after fixing this.**

So far, about three users have reported that the left half of their keyboard sometimes stops working all of a sudden. This is already a significant number and I expect it to be higher. Usability-wise this is clearly unacceptable, making this the most pressing firmware issue. Luckily, all evidence suggests that this issue can be fixed on the firmware level.

According to my findings, the KL03 microcontroller of the left keyboard half sometimes halts the I2C communication. Unplugging the left half and replugging it (hence power-cycling it) temporarily solves the issue, proving that the left keyboard half is the culprit. Furthermore, hardware-resetting the KL03 (by shorting the GND and RESET pins) in specific also always fixed the issue for me, which proves that the KL03 is the culprit (and not the left-side LED driver IC).

My planned solution is to implement an I2C watchdog for the left side firmware.

Interestingly, this issue never occurs on my current UHK, but I have a couple UHKs around, and I'll try to reproduce it to be able to implement and test a fix.

mondalaci, 2017-12-23:
@santiagogf89 I did implement a watchdog for the left half a while ago, but it broke the keyboard, and my UHK worked anyways, so I didn't think that it's important. Now it's clearly important, and I'd like to make it work.

In order to activate the the watchdog:
1. Uncomment `prevWatchdogCounter` and `I2C_WATCHDOG_LPTMR_HANDLER()` in [left/src/i2c_watchdog.c](https://github.com/UltimateHackingKeyboard/firmware/blob/1ac637af0f7eab867ef29bd1d58b05a920fd6f84/left/src/i2c_watchdog.c)
2. Uncomment [InitI2cWatchdog()](https://github.com/UltimateHackingKeyboard/firmware/blob/1ac637af0f7eab867ef29bd1d58b05a920fd6f84/left/src/init_peripherals.c#L88) in init_peripherals.c

ErichStyger, 2017-12-24:
Maybe I'm missing something, but: this is not a watchdog implementation, but instead uses the LPTMR. Is there a reason why the hardware watchdog of the KL03Z is not used?

ErichStyger, 2017-12-24:
I had to move the initialization of the timer after the I2C setup (otherwise an hard fault can happen). Plus I delayed the timeout check after the first 100 ms to avoid possible race conditions.
I have quickly tested the implementation, and it looks it is working fine.
Pushed as https://github.com/UltimateHackingKeyboard/firmware/commit/5581dd26b576c4a1b489d217ccd642794db9823e

mondalaci, 2017-12-24:
@ErichStyger I didn't really consider using the hardware watchdog. When it comes to using the hardware watchdog, is COP the only option for the KL03? @santiagogf89 mentioned to me today that [it's useless](https://community.nxp.com/thread/457893) for us according to errata e8058 as we must jump to the ROM bootloader.

Unfortunately, your commit seems to broke the left keyboard half. I don't know what's going on, but it doesn't respond to keypresses at all. To reephasize my earlier finding, when I included a totally empty `I2C_WATCHDOG_LPTMR_HANDLER()` it similarly made the left keyboard half unresponsive.

TorC8, 2017-12-25:
I just received mine, and have had some issues that look related to ths.  My experience seems to suggest that there may be an issue with a partial disconnection of the pins, primarily when flexing the keboard down in the middle.  Plugging in the bridge cable solved all issues in my as yet brief testing.  Recommended test for this would be to set the keyboard up with feet on only the four very outside corners.

TorC8, 2017-12-25:
After further testing, I'm more sure this one is mechanical.  On my specimen there is a gap between the halves that appears to take up nearly the entire working length of the pins.  It's still tight enough that with a well-supported keyboard the connection seems reliable, but when supported on the ends it can drop.  I wouldn't   With the UHK in my lap I observed the display blinking out on occasion, and correlated with responsiveness issues.  I've attached a picture.  A little darker than ideal, perhaps, but I think it shows what I'm seeing well enough for now.
[image]

TorC8, 2017-12-25:
Well, I went ahead and tested what appeared to be the most likely piece preventing the halves from closing properly.  With the case removed (the easily replacable black/coloured case), I can't duplicate the issue, even wiith a substantial amount of force downward in the middle.  With the case back on, I'm finding that I have to push the halves together to make them work.  That, or just plug in the bridge cable.  That invariably works as far as I've seen.  I'll leave it at this until tomorrow.  At some point I'll probably try shaving the ends of the case, so let me know if you want any tests done before I do.

ErichStyger, 2017-12-25:
@mondalaci I have now updated the I2C watchdog timer. The issue was that the LPTMR was used both for the key scanning and for the watchdog timer. Now it works properly for my left half keyboard too. The watchdog function can now be disabled/enabled in main.h if needed.

mondalaci, 2017-12-26:
@TorC8 Thank you for reporting your issue! We're fairly sure that yours is mechanical in nature and not firmware related. I've just shoot you an email. Please let's discuss your issue separately in email to keep this issue on-topic.

mondalaci, 2017-12-26:
@ErichStyger I'm humbled and ashamed! I implemented the LPTMR-based keyboard scanning routine a while ago, and totally forgot about it. I could have easily fixed this myself if I remembered, but instead I dragged you here in the middle of the holidays. I'm sorry about this. Thanks so much for your help!

mondalaci, 2017-12-26:
Everybody, If you experience this issue, please download [Agent 1.0.2](https://github.com/UltimateHackingKeyboard/agent/releases/tag/v1.0.2) and upgrade the bundled firmware within Agent. Chances are good it'll fix this issue, but I'm not sure. I'm waiting for your reports.

mondalaci, 2017-12-26:
Update: This fix does not seem to be working. Please don't update to the latest firmware! Unlike the UHKs of a couple of backers, mine was rock solid, and the left keyboard just hung after about a hour of use with the new firmware. I'm still trying to fix this. Will be keeping you up-to-date.

Hurricaaane, 2017-12-26:
I just wanted to note that in my case, the issue only happened once in the last 6 days (since I got the keyboard). My keyboard is connected through the bridge cable at all times.

opie4624, 2017-12-26:
Is it possible this is related to voltage supplied to the keyboard? I have it happen almost constantly when I'm connected directly to my computer, and never at all when connected through a powered hub.

When I look at the voltage on a scope, the power from my computer looks "dirty". There's lots of noise and power spikes, frequently up to 5.5v, which seems to coincide with my left half missing a keypress. Less frequently, I see spikes higher than that, which seems to coincide with the left half failing to respond altogether.

mondalaci, 2017-12-26:
@opie4624 Yours is a very interesting finding!

I always use my UHK via a powered hub and it's always rock stable which reinforces your supposition. From this point on, I'll connected my UHK directly to my PC to see if it makes any difference.

1. Would you elaborate on the specs of your computer?
2. When you say that it happens almost constantly, what do you mean? How frequently exactly?
3. How often does your keyboard hang if it's directly connected to your computer, and the halves are connected via the pogo pins instead of the bridge cable?
4. Can you see these power spikes when a) the left half is disconnected, and b) when the left half is connected via the pogo pins?

opie4624, 2017-12-26:
1. Would you elaborate on the specs of your computer?

  * Dell XPS 13 (The old original "Comes with Ubuntu" laptop)

2. When you say that it happens almost constantly, what do you mean? How frequently exactly?

  * I can't type my login password because left-half keys are missed about every third keystroke.
  * Left half locks up completely about every 5-10mins

3. How often does your keyboard hang if it's directly connected to your computer, and the halves are connected via the pogo pins instead of the bridge cable?

  * ~once a day _maybe_

4. Can you see these power spikes when a) the left half is disconnected, and b) when the left half is connected via the pogo pins, and c) (my add) the keyboard isn't connected at all?

  * Yes to all 3. The keyboard doesn't seem to do anything but pull down the voltage by ~.02V (.01v per half)

opie4624, 2017-12-26:
Scope screenshots: https://imgur.com/a/sNX4t

NB: I'm a n00b with an oscope, so do LMK if I'm chasing ghosts or I should be setting it up differently.

mondalaci, 2017-12-26:
In all honesty, I'm hopeless when it comes to analog electronics, but your scope pictures seem interesting, and I'll ask around.

Do you always use the same USB cable for connecting your UHK to hosts? Is it the one we provided?

opie4624, 2017-12-27:
I've been using the same cable that was provided. I will try with another, but the voltage issue I see seems limited to my one system. My Macbook is nice and steady.

mondalaci, 2017-12-28:
I've built a couple of new firmware images which hopefully improve this issue.

First up, please install the newly released [Agent 1.0.3](https://github.com/UltimateHackingKeyboard/agent/releases/tag/v1.0.3)

Then download the firmware images:

* [uhk-firmware-8.0.1-led-drivers-disabled.tar.bz2](https://github.com/UltimateHackingKeyboard/firmware/releases/download/8.0.1/uhk-firmware-8.0.1-led-drivers-disabled.tar.bz2) - This image has the LED driver ICs disabled (the LED display will be off) which may reduce the electrical noise, hence improve stability. Of course, this is not supposed to be a permament solution.
* [uhk-firmware-8.0.1-watchdog-reinit.tar.bz2](https://github.com/UltimateHackingKeyboard/firmware/releases/download/8.0.1/uhk-firmware-8.0.1-watchdog-reinit.tar.bz2) - This has a watchdog implemented in the left keyboard half that should reinitialize the communication between the halves if they go wrong.
* [uhk-firmware-8.0.1-watchdog-reboot.tar.bz2](https://github.com/UltimateHackingKeyboard/firmware/releases/download/8.0.1/uhk-firmware-8.0.1-watchdog-reboot.tar.bz2) - Very similar to the previous one, but more brute force. Instead of reinitializing communication, this reboots the left microcontroller upon communication errors.

Finally, flash the images one by one via the new Agent:

[image]

Please don't disconnect your keyboard during the update. If your keyboard display shows "FTY" after the update then power cycle your keyboard. You may also have to restart Agent after the update if it doesn't work.

Please let me know about your findings!

mikhey, 2017-12-29:
TL;DR:
This problem seems specific to Windows in my experience.

When I received my UHK yesterday, I connected it up to my personal MacBook Pro and it worked like a charm, no problems at all - both sides work w/o issue.  

I took my UHK to the office where I have a Dell 4800 Precision laptop as my primary workstation.  I connected it up to one of the Apple Cinema HD Displays and it worked for roughly 10-15 min.  I thought I had connected the top cable incorrectly, so I adjusted it and pulled the USB and reconnected it and it worked fine, again for a time.  After the 3rd time this occurred, I connected the UHK directly into a USB port on the laptop.  It worked fine for longer, but then at the end of the day, I started getting the same issue(s).   

I had used the Agent on the PC to attempt to load the watchdog-reinit firmware and I thought I had bricked it.  I connected it to my MacBook Pro and I was able to fully load the default firmware.

I'm wondering if there is something that Windows is expecting (or not expecting?).  My gut is it may be related to some kind of "polling" and I'm definitely not a hardware developer.  That being said, is there any tools or more information that I can provide to assist in troubleshooting the issue?

I'll take a look in the Windows logs tomorrow and post some of that information.  

All in all though, I <3 this thing!

mondalaci, 2017-12-29:
@mikhey Thanks for the info! It's very useful.

Would you please give every one of the 3 firmwares a try, (possibly by updating your UHK on your MacBook Pro) and see which ones solve the issue (if any)?

Unfortunately, I don't know of any way to debug this without special gear.

spuder, 2017-12-29:
For those experiencing this issue, consider if any of these other variables
may apply.

## Noisy power

Does it happen both when your laptop is on battery power and when plugged
into a wall?
Does it happen when UHK is plugged into a USB hub

## RF interference

Does it happen in other locations?
Are there wireless access points, microwave ovens, wireless keyboards mice
in proximity?
Are the halves separated or connected?

## Electrostatic discharge

the screws and bottom plate of the UHK are connected to ground.
Is the UHK placed on a conductive surface?
Are you in dry environment where there may be a lot of static build up and
discharge through the UHK?

## Other triggers

Does it happen after a certain amount of time?
Does it happen after a certain number of key presses?
Does it happen right when you sit down at your desk (static discharge)?
Does it happen after computer wakes from sleep?

spuder, 2017-12-29:
For those experiencing this issue, consider if any of these other variables may apply. 

## Noisy power 

Test when your laptop is on battery power and when plugged into a wall.
Try plugging the UHK is plugged into a USB hub

## RF interference

Try other locations (work, home)
Check for wireless access points, microwave ovens, wireless keyboards mice in proximity
Try using a [shielded 4P4C RJ22 cable](https://ultimatehackingkeyboard.com/faq#replacing-the-bridge-cable)

## Electrostatic discharge 

The screws and bottom plate of the UHK are connected to ground. 
Is the UHK placed on a conductive surface? 
Are you in dry environment where there may be a lot of static build up and discharge through the UHK? 
Try covering the screws and other metal parts in electrical tape

## Other triggers 

Does it happen after a certain amount of time?
Does it happen after a certain number of key presses?
Does it happen right when you sit down at your desk (static discharge)?
Does it happen after computer wakes from sleep?

---

I'm finding when I walk across carpet and sit down at my desk that the keyboard is rebooting/locking up. Most likely related to static discharge.

mikhey, 2017-12-29:
Okay, I just tried the LED disabled and watchdog re-init firmware - no luck.  With the LED disabled I couldn't even login to my Windows machine.  The re-init firmware didn't work, it eventually died and I had to re-plugin the UHK.  I'm using the watchdog-reboot firmware now and it seems to keep it going although I get the occasional missed letters on the left side of they keyboard.  In addition, since the watchdog reboot the LCD is no longer displaying.  It's an improvement in that I can a least type with both halves of the keyboard.

Interestingly an ALPS Touch Pad diagnostics window no pops-up when I plug-in the UHK.  Here's a screenshot (I just discovered it happens because the control key is held when connecting the device):
[image]

It's so odd that it does this on Windows, yet it works w/o issue on the Mac.  Also, I'm not seeing anything in the Windows logs.

mondalaci, 2017-12-30:
Thanks for the feedback and suggestions, guys! Please keep them coming!

Some tips:
1. As @spuder correctly noted, try to use a (preferably shielded) [4P4C crossover cable](https://ultimatehackingkeyboard.com/faq#replacing-the-bridge-cable). Instead of a spiral cable, a regular cable may work better. If you're experiencing this issue then we're happy to reimburse this expense of yours.
2. Temporarily tape down the pogo pin connectors of the left and right halves to see whether it solves the issue.
3. Temporarily remove the case of the keyboard, place it on a non-conductive surface and give it a try. We'll provide a disassembly guide soon.

Right now, I have 4 (soon 6) UHKs in my room connected to different machines running 24/7. I'm trying hard to reproduce this, but I just cannot seem to freeze these darn things.

opie4624, 2017-12-30:
Given what I'm seeing on the scope when I switch between my Macbook (Working) and my Dell/Linux (Bugged), the only difference is USB line voltage.

On the Dell, with _any_ device connected to the port, the voltage output ranges from 5.02v-5.55v, with the average being 5.2v.

On the Mac, with _any_ device connected, the voltage output ranges from 4.82v-5.22v, averaging 5.03v. (The same is true with most powered USB hubs I have access to.)

My suspection is that the increased voltage, which is out of USB spec, is causing the whole keyboard (but manifesting in flakiness in the left half) to generally error more often. Again, the voltage spread and average are the only things I can definitively point to as being different between the environments where the keyboard works flawlessly, and those where it has issues.

All 3 test firmwares behave the same, though the reinit and reboot versions recover themselves.

mondalaci, 2017-12-30:
@opie4624 Very interesting finding! On my computers, voltage is within spec (4.8V-5.25V). Somewhat contradictory though is that @spuder also uses a Macbook and his UHK hangs from time to time, so the physical environment may also be a factor to consider.

mondalaci, 2017-12-30:
You brought your laptops, your UHK, and your portable oscilloscope with you and made measurements everywhere? Now that's what I call true dedication! :) :+1:

I'm glad to hear that at least your results are consistent regarless of the environment. I plan to make a tour in Media Markt (similar to Micro Center and Best Buy) in the hope of finding a laptop on which it's consistently reproducible.

mondalaci, 2018-01-03:
Looks like @spuder has figured out the root cause of this issue, guys. It's the bridge cable! It likely picks up noise from the environment and makes communication unstable in some environments, making some UHKs hang over time.

If you're affected, would you please test this? Merge the keyboard halves, remove the bridge cable, then leave your keyboard on for a while. It shouldn't hang this way. Please report back with your findings!

Luckily, this can be vastly improved on the firmware level. I have a 20 meter long bridge at my hands with which I can reproduce this very easily, so I'm finally able to fix the issue, and actually test my fix.

I'll be keeping you updated.

mondalaci, 2018-01-17:
Long time no talk! Sorry for taking this so long guys, but hopefully the following will resolve this issue.

First up, let me put the affected users into one of two groups:

* You're **mildly affected** if your UHK freezes every now and then (on a weekly or daily basis, or possibly in every couple of hours), but other than that, your UHK is fully usable. According to your feedback, the vast majority belongs to this group.
* You're **heavily affected** if the above happens to you, and also your left keyboard half noticably ignores keypresses from time to time. As far as I know, only @spuder and @opie4624 belong to this group, and even in their case this only happens with specific computers at specific locations.

According to my investigation, the root cause of the freezes is the high parasitic capacitance of the I2C bus which interconnects the two keyboard halves and the add-on modules. High parasitic capacitance causes the signal shape to degrade, and ends up looking like this:

[image]

Versus the following signal which much more resembles an ideal square shape, therefore it's much more robust:

[image]

The signal further degrades when using a longer cable than the default one. My investigations showed no correlation between shielding and the shape of the signal, so noise probably doesn't affect the signal, although it may be possible.

I was able to easily reproduce the freezes with 16+m long cables. What I found is that the microcontroller of the left half eventually freezes, presumably due to the deformed I2C signal shape. Then the I2C transceiver of the right microcontroller also hangs, along with every other interrupt handlers, including the keyboard matrix scanner timer interrupt, and the USB interrupt, ultimately making the UHK detach from USB.

**Solution 1**: Flash the latest [8.1.0 firmware](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v8.1.0) with Agent. This has the previously referred left half "watchdog reinit" code enabled. This causes the left half to recover, which also makes the right half resume upon occasional freezes, so your UHK should keep going on forever. I received a report from a user (hi Börje!) that this solution completely fixed freezes for him. I also gave it a try, and I could confirm that my UHK always recovered.

This should fully resolve the issues for mildly affected users. If you're heavily affected, please read on.

**Solution 2**: Try another bridge cable. See some information on [replacing the bridge cable](https://ultimatehackingkeyboard.com/faq#replacing-the-bridge-cable). The shorter cable you pick, the more reliable your UHK should be. You don't have to use a spiral cable. As a matter of fact, a straight cable may be a better option. Repurposing a USB cable by crimping 4P4C connectors to its ends should work pretty well as USB cables are typically shielded (not sure if it matters, but certainly shouldn't hurt) and widely available.

**Solution 3**: Reduce I2C speed. First up, flash the latest firmware (Solution 1), then build the latest Agent. Then run `packages/usb/get-i2c-health.js` which should output something like the following right after powering your keyboard:

```
uptime: 0d 00:00:06
requestedBaudRate:100000 | actualBaudRate:104166 | I2C0_F:0b00101100
leftHalf      : nak:162 
leftAddon     : nak:6482 
rightAddon    : nak:6490 
rightLedDriver: 
leftLedDriver : 
kboot         : 
```

Uptime shows how long your UHK has been powered. Then you can see the requested I2C baud rate which is 100k by default. The actual baud rate somewhat differs because the I2C frequency register is only 8 bit wide. You shouldn't care about the register value but it correlates with the actual baud rate.

Then you can see the I2C error counters on a per I2C slave basis. nak means that the slave couldn't be reached. The counter will keep incrementing for the left and right add-on because they're not connected. The initial ~162 naks of the left half is normal because the microcontroller of the left half needs some time to initialize and the right half can't talk to it until it happens.

The interesting thing is how these values change over time. This is what happens under normal circumstances with the default bridge cable over the course of 12 minutes:

```
uptime: 0d 00:11:49
requestedBaudRate:100000 | actualBaudRate:104166 | I2C0_F:0b00101100
leftHalf      : nak:162 arbitrationLost:1 
leftAddon     : nak:726072 timeout:1 
rightAddon    : nak:726080 
rightLedDriver: 
leftLedDriver : 
kboot         : 
```

The above clearly belongs into the "mildly affected" category. You can see an arbitration lost error and a timeout, but nothing else. Such errors can lead to occasional freezes over a long time, but thanks to the "watchdog reinit" fix, such freezes shouldn't happen anymore.

Now see what happens with my 16m long bridge cable over the course of just one minute:

```
uptime: 0d 00:01:01
requestedBaudRate:100000 | actualBaudRate:104166 | I2C0_F:0b00101100
leftHalf      : nak:102 arbitrationLost:23 busy:8 timeout:119 
leftAddon     : nak:23487 timeout:23 busy:119 
rightAddon    : nak:23493 
rightLedDriver: 
leftLedDriver : 
kboot         : 
```

You can see loads of I2C errors of all kinds, including timeouts, busy states, and arbitration lost events. (Interestingly many of these belong to the left add-on which is clearly disconnected. These errors probably shouldn't be counted on a per slave basis because parasitic capacitance related I2C errors make a [... cut ...]
Just to get things straight, such modifications will not void your warranty. This issue is clearly our fault, and we're super grateful to you for helping to fix it. We're also happy to reimburse any relevant cost, such as the price of alternate bridge cables. Just let us know.

And that is it! Thanks for your continued patience, everybody! I think we're well on our way to make the UHK rock stable.

**Every one of you who is affected, please do provide feedback about this issue! It's critical to get this right in the future.**

mikhey, 2018-01-29:
I'll try this firmware out.  Weather hasn't been so great lately and I haven't been bringing from home to the office!  Thanks for your efforts to look into it!

mondalaci, 2018-02-11:
According to the recent feedback we received this seems to be resolved. If you experience any issues, feel free to comment, and it'll be reopened if needed.

closed 2018-02-11 by mondalaci

spuder, 2018-11-03:
These resistors work https://www.mouser.com/ProductDetail/754-RR1220P-102D

Video of how to swap those resistors

mondalaci, 2018-11-03:
Yep, these look great.

Are you looking for a video on how to swap the resistors?

mondalaci, 2023-11-07:
@ration Please see [this troubleshooting guide](https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys).

ration, 2023-11-08:
Thanks, I lowered the led brightness and it hasn't happened again (yet). Fingers crossed!

jcdupuis-ulaval, 2024-09-20:
I am experiencing a similar issue but in my case I think that it comes from the physical connection of my Trackpoint Module. When vertical force is applied to depress the buttons or to push on the Trackpoint the module disappears in UHK Agent and so does the left hand side of the keyboard. The problem disaopears if I hold the module steady with the other hand. The watchdog implemented in the firmware appears to do its job and the left side of the keyboard and Trackpoint Module reappear eventually. The cantilever design of the module appears to be causing intermittent signal degradation.