UltimateHackingKeyboard/firmware #75: The sleep key is not working

mondalaci, 2017-12-25:
The sleep key is only working on Linux. It's not working on any other OSes. The system keyboard USB report doesn't seem to be right. Fixing that should probably fix this issue.

Lauszus, 2018-02-11:
This might only be supported on Linux: https://github.com/UltimateHackingKeyboard/firmware/blob/master/right/src/arduino_hid/ConsumerAPI.h#L28.

Lauszus, 2018-02-11:
I got it working on Mac by sending the ```CONSUMER_SLEEP``` key instead. I'll test this on Windows and Linux tomorrow.

mondalaci, 2018-02-12:
Thanks! The sleep key should work on Windows. I tested the sleep key of the ultra-cheap Hama Media Keyboard "AK 220" on Windows 10 Ultimate and it did work.

Lauszus, 2018-02-13:
#93 fixes this issue on Windows. However you need to send the consumer key on Mac instead. My recommendation would simply be to map the sleep key to ```SYSTEM_SLEEP``` on Windows and ```HID_CONSUMER_SLEEP``` on Mac.

mondalaci, 2018-02-14:
@Lauszus Just tested this, and here are my findings:

* Linux respects both system sleep and consumer sleep.
* Windows 7, 8.1 and 10 respects system sleep but not consumer sleep.
* OSX 10.9.5 respects neither system sleep nor consumer sleep.

Any idea about OSX? I couldn't find any info on it. Maybe it's simply not possible to put it into sleep via USB.

Lauszus, 2018-02-14:
Consumer sleep works fine on my Mac running 10.13.2.

mondalaci, 2018-02-14:
Interesting. Maybe newest OSX versions behave differently in this respect than older versions. [These shortcuts](https://www.cultofmac.com/269328/handy-keyboard-shortcuts-shut-sleep-mac-os-x-tips/) however work for me.

How about the Power Down USB key?

Lauszus, 2018-03-10:
@mondalaci those shortcuts works fine with me as well. Maybe we could add a macro for them?

No ```SYSTEM_POWER_DOWN``` does not work for me.

mondalaci, 2018-03-11:
@Lauszus Actually, macros are not needed, merely shortcuts. I agree that there doesn't seem to be any other way for Macs.

Lauszus, 2018-03-11:
@mondalaci what I meant by macro was simply that we should sent out multiple scancodes if the "sleep button" was pressed.

mondalaci, 2018-03-30:
I've looked into this further, and these are my findings:

On Linux:
* The Sleep scancode puts the PC to sleep.
* The Power Down scancode makes the shut down dialog pop up.
* The Wake Up scancode puts the PC to sleep, then makes the shut down dialog pop up after wake. (The latter making no sense.)

Windows 7, 8.1, 10 does not differentiate between these scancodes. Each of them puts it to sleep.

OSX 10.9.5 does not differentiate between these scancodes either. When tapping them nothing happens, but when one of them is held for 2 secs the shut down dialog pops up.

As for the aforementioned [OSX shortcuts](https://www.cultofmac.com/269328/handy-keyboard-shortcuts-shut-sleep-mac-os-x-tips/), they seem to only work if the modifiers are held before sending out the scancode itself. This makes the UHK shortcut functionality not work in this case. Macros would work as suggested, but they're not implemented yet.

Closing this now, as the UHK implements sleep as USB keyboards should. We'll work around the quirkiness of OSX once macros get implemented.

PR #93 (2018-02-13): Fix system keyboard descriptor, so it is byte-aligned
I have verified that the descriptor shows up correctly on Windows 7 and macOS.

The descriptor is based on the two following pages: https://github.com/Microsoft/Windows-driver-samples/tree/master/hid/hidusbfx2 and https://docs.microsoft.com/en-us/windows-hardware/drivers/gpiobtn/hid-button-report-descriptors.

Fixes #76.

I believe this fixes #66 as well.

From my testing this fixes #75 and #80 on Windows.
closed 2018-03-30 by mondalaci