UltimateHackingKeyboard/firmware #78: Mac mouse should be inverted

spuder, 2017-12-28:
OSX by default does inverted scrolling. (Since OSX Lion)

The `Y` and `H` scrolling feel wrong to me. I think if you have selected QWM, it should default to inverted scrolling on Mac.

Two ways to querry the OS setting for scroll

https://stackoverflow.com/questions/30767674/change-scroll-direction-on-mac-programatically-assign-to-key-combination-comman

https://stackoverflow.com/questions/12433018/how-to-check-scrolling-direction-of-osx-with-cocoa-apis

spuder, 2017-12-28:
Never mind, I now realize that the Y and H key are just doing a 'page up' and 'page down'. The UHK should be consistent with those keys.

closed 2017-12-28 by spuder

mondalaci, 2017-12-28:
I assume you closed this due to https://github.com/UltimateHackingKeyboard/agent/issues/535

Looks like we're totally on the same page regarding the solution. Thanks for reporting!

spuder, 2017-12-28:
Yes, it is a duplicate.