UltimateHackingKeyboard/firmware #79: Surprising behavior of layer switch locks

TorC8, 2017-12-28:
The layer switch locks on double press of layer switch key are a nice feature (clearer documention of it might be nice at some point), but I've found they also can be a little surprising.  I would suggest that the lock should be activated only on a double quick press, but if the second press lasts more than ~.5s then assume the user was careless about initially pressing the key (or something went wrong with switch debouncing) and release the layer when the layer switch key is released.

Hurricaaane, 2017-12-29:
I agree with this. Sometimes when I want to call the mod layer, I accidentally tap the mod key once and then then hold it.

I like the locking mechanism but I don't think it should remain locked if at least one key was pressed while a layer key is being pressed.

mondalaci, 2017-12-29:
Excellent idea! Wouldn't have though of this by myself. 0.5s should be a reasonable value, and maybe it'd be worth to make it customizable.

(The 0.25s delay between the taps is already customizable on the firmware level, but yet to be exposed via Agent.)

Hurricaaane, 2018-02-10:
I wanted to add that since I got the keyboard, this has probably been the biggest gripe I had. Several times a day while programming, my keystrokes will suddenly start to do insane stuff in the IDE and I'll stop and look down at the keyboard to find out whether it's the Fn or the Mod key that is locked.

I'll probably disable the double tap completely when the option comes, unless the new implementation of the double tap is really good. I'd rather like to configure a specific keystroke in the agent to lock a key, like Fn+Backslash to lock the Fn layer or something similar.

closed 2018-02-11 by mondalaci

mondalaci, 2018-02-11:
@Hurricaaane I agree that this was the most irritating bug of all usability-wise. Just fixed it. Please flash [firmware 8.1.1](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v8.1.1) with Agent.

I set the release timeout to 100ms which I believe is an optimal value. Eventually, I'd like this to be customizable via Agent. I've already closed this issue, but feel free to report your findings.

MattKunze, 2018-06-18:
> Eventually, I'd like this to be customizable via Agent

Definitely would like to see this customizable as well. I think I'm in a similar situation to @Hurricaaane where I'd like it essentially disabled. The only times it gets locked for me are false positives (this doesn't happen all that often to be fair though)

mondalaci, 2018-06-18:
@MattKunze Gotcha. This is partially implemented and should be resolved soon.

commit 2c9a6c0b (2018-02-11): Add doubleTapSwitchLayerReleaseTimeout and set it to 100ms. Resolves #79.