UltimateHackingKeyboard/firmware #82: Activating and dropping a layer drops held modifier keys

TorC8, 2017-12-29:
This occurs even when the modifier is defined as such in both the base layer and the relevant layer.  For example, hold Alt, hit tab to bring  up the window switch list, then pres and release Mod.  Despite the key being defined as Alt on both Base and Mod layers, when releasing Mod the modifier is dropped.  I've proven this to occur with Shift, Control, and Alt.

As a note, I think the correct behavior would probably be to maintain the modifier until the key is physically released even if the layer has the key defined differently, in effect masking the key on the newly activated layer, unless the key is first released.  While this would prevent the use of the layer-specific keycode with the modifier, I think this would probably be the least surprising beavior overall.

mondalaci, 2017-12-30:
Great catch! I could reproduce this bug and agree about the suggested solution.

bittrance, 2018-05-02:
I'm not 100% sure, but I think I have a variant of this issue. Specifically, in Atom I select text (`shift` + `alt` + `mod` + `l`) and then I want to add parens around the selected text which Atom lets you do by simply pressing start paren (i.e. `shift` + `9`). However, since the shift modifier is forgotten, the text is replaced by a "9". Rather annoying. OP suggested solution would handle this case.

commit 14b6e496 (2018-05-02): Don't suppress modifiers upon releasing a layer. Fixes #82.
closed 2018-05-02 by mondalaci