UltimateHackingKeyboard/firmware #83: Disable display during sleep

mondalaci, 2017-12-31:
To save power.

borje, 2017-12-31:
I agree to this.

Lauszus, 2018-02-13:
I had a quick look at the document "Hut1_12v2.pdf" which defines all the HID Usage Tables. See page 61 chapter 11. 

It looks like we can simply extend the LED usage page: https://github.com/UltimateHackingKeyboard/firmware/blob/fa8c9dc907ef80a21ddcc6002175f1a0c4c42351/right/src/usb_descriptors/usb_descriptor_basic_keyboard_report.c, so it includes "Power" (id 6) or "Stand-by" (id 27).

We have to make sure it still is hidboot compliant, but I believe at least "Power" should be.

mondalaci, 2018-02-14:
Thanks for looking into it! This sounds like the way forward when we get there. Currently it's not a high priority issue.

PR #97 (2018-03-11): Turn LEDs, backlight and display off when sleeping
Fixes #83 

There is still a problem with waking up the computer on macOS again. Please test this on Linux and Windows.
commit 5de0e5ac (2018-03-11): Turn LEDs, backlight and display off when sleeping

Fixes #83
closed 2018-04-10 by mondalaci