UltimateHackingKeyboard/firmware #84: Whenever the keyboard is plugged in a host machine, Chrome Remote Desktop is blocked from remotely interacting with the machine

Hurricaaane, 2017-12-31:
Whenever the keyboard is plugged in a host machine, Chrome Remote Desktop is blocked from remotely interacting with the machine.

Let's say I have two machines:
- A desktop with UHK plugged in
- A Chromebook

If I use my Chromebook to connect to my desktop machine remotely using Chrome Remote Desktop,
- The Chromebook will not be able to take control of the mouse and keyboard if the UHK is plugged in
- When the UHK is unplugged while the Remote Desktop session is running, it will allow the Chromebook to take control of the mouse
- When the UHK is plugged again, it will prevent the Chromebook from taking control of the mouse again

Usually, Chrome Remote Desktop prevents the Chromebook from sending inputs to the desktop if the host's keyboard has recently been typed on when the host's mouse has recently been moved. Presumably the UHK sends some kind of events that triggers some condition?

Anyways it means that the UHK is not behaving like a traditional device. This is a huge problem for me as I don't want to unplug the UHK whenever I leave home.

mondalaci, 2017-12-31:
This is rather surprising, because the UHK doesn't do anything super-fancy on the USB level that gaming keyboards / mice don't do.

My assumption is that your desktop OS is not in peace with the descriptor of some of the USB interfaces of the UHK, so the descriptor must be further tweaked on the firmware level.

May I know the OS of your desktop machine?

Hurricaaane, 2017-12-31:
The desktop machine is Windows 10.

To add to the issue, I've noted when I plug my keyboard, a peripheral appears in my list of peripherals under the category "Mouse and other pointing devices" (translated from French), HID Mouse. If keep my UHK plugged in and disable this peripheral, it no longer blocks the remote desktop inputs (but obviously I lose the ability to use the UHK mouse functionnality).

If you need more details don't hesitate to ask me.

mondalaci, 2017-12-31:
Thanks for the useful info! Will let you know if I need more help.

mondalaci, 2018-02-12:
I'd like to add that according to my testing this only happens if the device being shared runs Windows. I guess the other end of the connection doesn't matter, but I used Linux.

This should be resolved in the near future now that we have a USB expert on our team.

closed 2018-06-07 by mondalaci

Hurricaaane, 2018-06-10:
It works! Thanks ✔️

PR #120 (2018-06-07): The UHK now only sends out data when it changes
Fixes #72, fixes #84 and fixes #85
commit 10985abc (2018-06-07): The UHK now only sends out data when it changes

Fixes #72, fixes #84 and fixes #85