UltimateHackingKeyboard/firmware #85: Windows: UHK prevents sleep

laxu, 2018-01-01:
With the UHK connected, my computer running Windows 10 64-bit won't go to sleep at all. Display won't sleep either so I assume that UHK keeps doing something over the USB as disconnecting it fixes the issue.

mondalaci, 2018-01-01:
I can reproduce this issue. I think chances are good it's related to #84.

laxu, 2018-01-02:
DIsabling the mouse in Device Manager doesn't seem to fix the issue for me.

Hurricaaane, 2018-01-02:
Same here, I can also reproduce this issue, but unlike #84, disabling the mouse the does not fix it.

Hurricaaane, 2018-01-08:
I noticed today that my computer went to sleep while my UHK was plugged in. Wondering why this was the case, I then noticed that the left half of the keyboard was not responding ( see #74 ). Unplugging the bridge cable and replugging it woke up my computer...

mondalaci, 2018-01-08:
@Hurricaaane Thanks for the info! #74 apparently affects the USB interrupt, too. Fixing #74 will make this always an active issue, so this will have to be separately fixed regardless.

laxu, 2018-01-09:
Tested this but alas no luck on my machine. I set Windows display sleep to 1 min and sleep to 2 min but my UHK has to be disconnected for either to happen. I tried disconnecting left side, disconnecting cable entirely and connecting the halves together without cable. So I think for this issue you can at least partially rule out the left side causing this.

mondalaci, 2018-01-09:
Disconnecting the left half doesn't eliminate this issue. It gets only temporarily eliminated if the communication between the halves halt due to #74. I think I don't have any more information for this to resolve it.

Jopie01, 2018-01-12:
To me this seems like there is some unsolicited communication from the UHK to windows. Once I had an Arduino Leonardo, which also could act as mouse. To cheat a bit, I created a program to moves the mouse a little bit every now and then, to show my colleagues that I'm still behind my desk and hard at work 😆. My system didn't lock itself or did go to sleep.

I'm also curious if this happens on other OS'es too. IMO it should behave the same. I will check on my system with Fedora 27

Jopie01, 2018-01-12:
I tested Fedora 27 and Windows 8 on my dual boot system.

- Fedora 27: No problems just works. Even sleep and blanking the screen. Dmesg after plugging in the keyboard gives shows me that there are 5?? input devices
```
[ 9697.381212] hid-generic 0003:1D50:6122.000D: hiddev97,hidraw2: USB HID v1.10 Device [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:12.1-1/input0
[ 9697.384163] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:12.1/usb4/4-1/4-1:1.1/0003:1D50:6122.000E/input/input21
[ 9697.437020] hid-generic 0003:1D50:6122.000E: input,hidraw3: USB HID v1.10 Keyboard [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:12.1-1/input1
[ 9697.441609] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:12.1/usb4/4-1/4-1:1.2/0003:1D50:6122.000F/input/input22
[ 9697.494297] hid-generic 0003:1D50:6122.000F: input,hidraw4: USB HID v1.10 Device [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:12.1-1/input2
[ 9697.498351] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:12.1/usb4/4-1/4-1:1.3/0003:1D50:6122.0010/input/input23
[ 9697.550704] hid-generic 0003:1D50:6122.0010: input,hidraw5: USB HID v1.10 Device [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:12.1-1/input3
[ 9697.555173] input: Ultimate Gadget Laboratories Ultimate Hacking Keyboard as /devices/pci0000:00/0000:00:12.1/usb4/4-1/4-1:1.4/0003:1D50:6122.0011/input/input24
[ 9697.556524] hid-generic 0003:1D50:6122.0011: input,hidraw6: USB HID v1.10 Mouse [Ultimate Gadget Laboratories Ultimate Hacking Keyboard] on usb-0000:00:12.1-1/input4
```
Also, `lsusb` show me that I have an OpenMoko Inc. device plugged in.... (ID 1d50:6122)
- Windows 8: No problem using the keyboard and mouse. Plugging in the keyboard gives me also 5 HID-devices in the device manager. One has a problem because there is a warning sign. Under section mouse, there appears a HID-mouse and under section there is a HID-keyboard.

On both systems I didn't have installed the Agent. Do I need extra drivers for the keyboard?

laxu, 2018-01-12:
I can confirm I also see the one device with a warning. Removing it has no effect on this issue.

I tried connecting my UHK to my 2016 Macbook Pro and display sleep at least works as normal. So this seems to be an issue with UHK and Windows. I tried checking `powercfg -requests` to see if there is any record there but it did not report any issues.

Jopie01, 2018-01-12:
> The UHK is a composite USB device which implements 5 USB interfaces, so this is normal.
OpenMoko allocated us several PIDs (instead of buying a VID from usb.org for $5K), so this is normal too.
We know about the problematic interface on Windows.
No extra drivers should be needed.

Thanks for clearing this up. I think that first #76 should be fixed before we can say anything more about this issue. 
**Just guessing**, maybe there are malformed messages going up and down from both systems (Windows and UHK are trying to make anything out of the messages?) which prevents Windows to go to sleep.

mondalaci, 2018-01-12:
No malformed messages should be going around. Windows just doesn't seem to be happy with the USB desciptor of an interface. I'll look into #76.

Lauszus, 2018-06-06:
I can confirm that the UHK sends data continuously via USB even though no buttons are pressed. Btw you can easily see the packets in real time using Wireshark: <https://www.wireshark.org/>.

Lauszus, 2018-06-06:
I just checked another keyboard and it does exactly the same thing.

Lauszus, 2018-06-06:
A third keyboard did the right thing and works with Windows. PR #120 should fix this issue. Please test and report back.

closed 2018-06-07 by mondalaci

laxu, 2018-06-07:
Can confirm, works now. I tested only the monitor sleep but since that works, so should the full sleep. Great work!

Hurricaaane, 2018-06-10:
Thanks, I also confirm it works ✔️

PR #120 (2018-06-07): The UHK now only sends out data when it changes
Fixes #72, fixes #84 and fixes #85
commit 10985abc (2018-06-07): The UHK now only sends out data when it changes

Fixes #72, fixes #84 and fixes #85