UltimateHackingKeyboard/firmware #86: Keymap switch instead of macro playback

mondalaci, 2018-01-02:
Macro playback is not supported yet, and macro play actions shouldn't do anything, but when the UHK is configured according to the picture below,
[image]
 for some reason Mod+Z switches to the test keymap instead of (not) playing a macro.

TorC8, 2018-01-02:
I can confirm this behavior.  Given that it's a macro I attempted to define, it appears to be consistently any macro playback attempt switches to the test keymap.  I have added a keymap, but not deleted any of the original kemaps.  One I did rename to a later-in-alphabet name and code, and added another earlier in alphabet.  I would wonder whether this is related to the test keymap or to its position in the keymaps array.

srang, 2018-01-04:
I'm seeing the same thing happen when I put a macro in the FN layer. I'll see if I can narrow that down at all to what key its being mapped on.

commit 66c877f7 (2018-05-14): Don't switch keymaps instead of playing macros. Fixes #86.
closed 2018-05-14 by mondalaci