UltimateHackingKeyboard/firmware #87: Bitlocker doesn't like the UHK

Jopie01, 2018-01-09:
My laptop has Bitlocker installed to encrypt the harddisk, so before I can start Windows, I have to enter the unlock-key. But just ticking one key generates a lot of (I think) characters in Bitlocker. I can't see it because it's a password-field.

Fortunately it's a laptop and I can use the keyboard of the laptop to enter my key. The UHK works fine in Windows (I didn't test it thoroughly though)

The laptop is a Dell Precision M4800 and the UHK is plugged directly into the USB-port of the laptop. It doesn't matter which USB-port I use. Next week I'm going to test this with the laptop in a docking station and the UHK also connected to one of the USB-ports of the dockingstation.

mondalaci, 2018-01-10:
This for the report, Jorge!

It's very strange, because I've just tried this, and no characters appeared
in the Bitlocker prompt when typing which is quite a different behavior
than yours. My UHK also worked fine in Windows. Please keep me updated
about this. Eventually an issue should be written about this in the
firmware repo.

Jopie01, 2018-01-12:
Even before Windows can be started, you have to decrypt your harddisk via Bitlocker. So after the BIOS Bitlocker is started to let you enter the encryption key. So I think this is a BIOS issue which does not recognize the keyboard correctly.

mondalaci, 2018-01-12:
I agree that this is probably BIOS-dependent. What's the type and version of your BIOS?

Jopie01, 2018-01-12:
> What's the type and version of your BIOS?

I have no idea. When I start the laptop the Dell logo is shown until Bitlocker shows up. Pressing keys doesn't change anything. Entering the BIOS gives me a graphical interface where I can select different options. The only thing about the BIOS itself is that it's version A18.

Can it be something with USB3.0 or secure boot?

laxu, 2018-01-12:
Have you tried if it works with a different external keyboard? On some slightly older desktop motherboards I've had issues with USB 3.0 ports that were using a non-Intel (VIA) controller. I had a board where only USB2 ports connected to an Intel controller worked in BIOS and the USB3 ports would only work after the 3rd party controller driver was loaded by Windows. If Bitlocker is initialized before this phase it might not accept input from external ports.

If your BIOS has a Fast Boot option, try disabling that and see if UHK works then.

Jopie01, 2018-01-12:
"normal" external keyboards give no problems, even when the laptop is connected with the docking station. The laptop is around 4 years old. 
My fastboot option is Thorough about which the help says:

> Performs complete hardware and configuration initialization during boot

Changing the BIOS also requires a special key which I don't have (only the IT-department)

mondalaci, 2018-01-12:
I'd be interested in disabling fastboot and giving this another try. Any chance you can ask help from the IT department?

Jopie01, 2018-01-12:
I only have 3 options:
- Auto : Allows the BIOS to decide configuration initialization performed during boot
- Thorough : (selected now)
- Minimal : Reduces boot time by skipping certain hardware and configuration initialization during boot

So which one do you want ... I have good relation with the IT-department, so I don't see no problem here.

mondalaci, 2018-01-12:
I think it'd be worth to try the Minimal option. I assume Auto picks either Thorough or Minimal.

laxu, 2018-01-13:
Minimal is actually the option more likely to cause problems as it should allow faster booting by reducing the number of device drivers it loads. Changing the fast boot option to disabled was a recommendation I saw when googling for Bitlocker issues with USB. Since other external keyboards work I would point this issue towards the UHK. Maybe it being a composite device causes some issues here or something.

Jopie01, 2018-01-24:
I think I'm hit by this https://www.ctrl.blog/entry/preboot-bitlocker-localization. So nothing I can do about it then file an issue with Microsoft.

Lauszus, 2018-04-10:
Is this still a problem after https://github.com/UltimateHackingKeyboard/firmware/issues/76 was fixed in https://github.com/UltimateHackingKeyboard/firmware/pull/93?

mondalaci, 2018-04-10:
@Lauszus It's not fixed, but I've just tested Bitlocker with 4 different keyboards, and it didn't work with either. As absurd as it may sound, I'm wondering if it supports USB HID devices at all.

@Jopie01 Would you give a couple of different USB keyboards a try using Bitlocker? I'm wondering if either of them works for you.

Jopie01, 2018-04-10:
I've tested it with several different USB keyboards, but only the UHK have problems. I think it may have to do with #72 because when I press a key I got 8 or 10 characters entered into the password field.

Another thing is, when I press the start button to start the laptop, and then plug in the UHK, the BIOS fails completely and goes into a kind of recovery mode. I have to plug in the UHK and then start the laptop.

mondalaci, 2018-04-11:
@Jopie01 Is the number of characters displayed dependent on how long you press a key? If so, I agree with your assumption.

As for the second issue, I don't have any ideas. It'd be worth to revisit it once related USB issues get fixed.

Jopie01, 2018-04-11:
> @Jopie01 Is the number of characters displayed dependent on how long you press a key?

Yes it does. One normal press gives you around 10 bullets. Pressing fast around 6 and pressing longer gives you a lot of bullets. For the time-being I still have a normal USB keyboard attached to my docking station and I have to do it just once, so I have a workaround.

mondalaci, 2018-06-07:
Just released [firmware 8.3.1](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v8.3.1). Please give it a whirl.

Jopie01, 2018-06-07:
I can confirm that firmware 8.3.1 fixes this issue. It's a joy to enter the PIN number now :smile:

mondalaci, 2018-06-07:
So glad to hear! I think it's safe to say that this bug was essentially the same as #72. Have fun! :)

closed 2018-06-07 by mondalaci