UltimateHackingKeyboard/firmware #89: Custom table-based layouts

LDVSOFT, 2018-01-24:
Hi there!

While reading through about your awesome keyboard, one of the keypoints was it's customization abilities. I am using on Linux desktop an alternative input method called somewhat-like `IBus table: LaTeX`, that allows user to type `\rightarrow` and get → just out of it, and a lot of other characters. I really like that ability, but setting up this method on every computer is quite compicated, and it is not available on Windows/Mac.

So, is there a chance to implement this feature _inside_ the keyboard as, for example, a special input layout? It would be really amazing:
* Your configuration is always with you,
* Input hints based on your usage are always up to date,
* Actually works everywhere!

I see several challenges implementing that:
* Displaying hints while typing: that may require agent-site support...
* Code/data memory requirements: but it looks like the keyboard has some pretty good chips, though...
* Keyboard must to be able to type Unicode sequences, that may be hard due to protocol limitations, but still, maybe with agent's help?

mondalaci, 2018-01-25:
Hi Dmitry,

What you're after should be (partially?) possible with macros. For example, you'll be able to bind Fn+L to make the UHK type `\rightarrow`. The macro engine is not ready yet, but it is planned to be implemented in the near future.

USB keyboards are only capable of sending scancodes, not characters, then the OS translates the scancodes to characters using the actual OS-speicfic layout. Even without `IBus table: LaTeX`, It's possible to bind keys like Fn+L to a sequence of [Alt-codes](https://en.wikipedia.org/wiki/Alt_code) implemented by macros which will result in unicode characters. Alt-codes are OS-dependent, but one way around it is to create multiple OS-specific keymaps in the UHK which will be stored in its memory and work with any OS without special software.

Interpreting keyboard input by the firmware in a special way while outputting relevant scancode sequences and displaying hints is another matter. That'd be a huge undertaking and a very niche application, and it'd also vastly complicate the configuration UI of Agent, so I don't foresee it ever implemented to be honest. Hopefully, the above features will work for you pretty well.

I'm closing this ticket, in light of the above, but feel free to follow up if you have any more questions.

closed 2018-01-25 by mondalaci

LDVSOFT, 2018-01-25:
Thank you for your view of the problem!