UltimateHackingKeyboard/firmware #90: Gamepad emulation mode

ata2001, 2018-02-03:
I'm working on adding a new `keyActionType` option to the firmware (and agent), which will make it possible for the UHK to act as a game controller. With this you'll be able to create a gamepad layer, much similar to the current mouse layer. Some split screen/local multiplayer games require a dedicated game controller for the second player to join the game, setting up two bindings on one keyboard or using two keyboards to play is not possible. With a gamepad emulation mode built into the keyboard firmware, this could be worked around.

There are two different standards for USB gamepads in use. Some gamepads function as standard HID game controllers, while other gamepads work the same way as the Xbox 360 controller, ignoring the HID standard. Linux supports both standards fine. On windows however HID gamepads get exposed as directinput devices. According to msdn, the DirectInput API is deprecated, and the new XInput API should be used for implementing game controller support in a windows application. Unfortunately the XInput driver in windows doesn't support HID gamepads. you either have to buy a new Xbox type gamepad or use workarounds like [x360ce](http://www.x360ce.com/) in order to play modern games.

I think going with the Xbox standard would make more sense. since it is well supported everywhere by now.

I found a [library](https://github.com/zlittell/MSF-XINPUT) implementing xinput, it was originally written to be used with the TeensyLC development board (arm). With some modifications we could use that.

Another problem is recognizing it as an XUSB compatible device. It seems like, the only possible way to identify such devices is by Vendorid/Productid (Vendorid/Interface SubClass may also work). Linux' xpad driver has a long array of known gamepads. http://elixir.free-electrons.com/linux/4.15/source/drivers/input/joystick/xpad.c#L128
I'm not sure, how it's done in Windows.

I could send a patch to xpad developers easily, but I have no clue, how could we get it to work on windows.

I own a Logitech F310 gamepad, that supports both gamepad standards, there's a switch on its back.

lsusb output:

Logitech F310 XInput mode:
```
Bus 005 Device 008: ID 046d:c21d Logitech, Inc. F310 Gamepad [XInput Mode]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc21d F310 Gamepad [XInput Mode]
  bcdDevice           40.14
  iManufacturer           1 Logitech
  iProduct                2 Gamepad F310
  iSerial                 3 XXXXXXXX
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           48
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     93 
      bInterfaceProtocol      1 
      iInterface              0 
      ** UNRECOGNIZED:  10 21 10 01 01 24 81 14 03 00 03 13 02 00 03 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
```

Logitech F310 DirectInput mode:
```
Bus 005 Device 009: ID 046d:c216 Logitech, Inc. Dual Action Gamepad
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc216 Dual Action Gamepad
  bcdDevice            4.14
  iManufacturer           1 Logitech
  iProduct                9 Logitech Dual Action
  iSerial                 3 XXXXXXXX
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      99
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
```

Todo:

Agent:

- [ ] Add gamepad keyActionType to serializer
- [ ] Add support to the GUI

firmware:

- [ ] Implement it in the config parser
- [ ] Create new interface descriptor
- [ ] Make it send game controller events

Links:
* http://free60.org/wiki/GamePad
* http://www.zlittell.com/2015/07/fightstick/

mondalaci, 2018-02-03:
> com/linux/4.15/source/drivers/input/joystick/xpad.c#L128

Thanks for the update! I really enjoyed it, and learned a ton from the
linked resources about how Xinput works.

I agree that Xinput as the way to go. It'd be hardly worth to implement
standard gamepad support just for legacy compatibility.

The Teensy library should be a good starting point, although it's vastly
different from the UHK firmware. Luckily, the USB descriptors of the UHK
firmware are exceptionally well organized.

The fact that Xinput is not HID based will add to the complexity of the
implementation. When I started to write the current UHK firmware, I based
it off of the usb_device_composite_hid_mouse_hid_keyboard demo
<https://github.com/UltimateHackingKeyboard/KSDK_2.0_MK22FN512xxx12/tree/master/boards/frdmk22f/usb_examples/usb_device_composite_hid_mouse_hid_keyboard/bm>
of KSDK, removed everything I didn't need, and added what I needed. The
remnant of this demo is the ksdk_usb
<https://github.com/UltimateHackingKeyboard/firmware/tree/master/right/src/ksdk_usb>
directory of the firmware. Given this background, all of the current UHK
interfaces are HID interfaces and ksdk_usb might have to be extended or
modified to deal with non-HID interfaces. It might be worth to check out
other KSDK USB examples
<https://github.com/UltimateHackingKeyboard/KSDK_2.0_MK22FN512xxx12/tree/master/boards/frdmk22f/usb_examples>
regarding how the KSDK deals with non-HID interfaces.

As for the VID/PID issue, I'm very surprised about the Linux
implementation. I guess we can just use one of these VID/PIDs for
development purposes and later we can get ours added to the list. I don't
have a clue about Windows (and OSX) either. We can figure it out later, but
again, for development purposes we can just use an already existing VID/PID.

I think it'd be a good strategy for you to focus on the USB and firmware
part of this now. You could simply temporarily repurpose a very rarely used
set of keys for Xbox gamepad buttons, for example F13-F24, then make the
firmware ignore these keys for the keyboard interface, and use them for the
newly created Xinput interface. Then you could configure them in Agent
without modifying Agent in any ways. Once you'll have the USB part working,
you can count on us to make this happen in Agent.

Keep it up, and please keep me posted! I'll try to help if needed.

mondalaci, 2018-02-04:
Thanks for the update! I really enjoyed it, and learned a ton from the linked resources about how Xinput works.

I agree that Xinput as the way to go. It'd be hardly worth to implement standard gamepad support just for legacy compatibility.

The Teensy library should be a good starting point, although it's vastly different from the UHK firmware. Luckily, the USB descriptors of the UHK firmware are exceptionally well organized.

The fact that Xinput is not HID based will add to the complexity of the implementation. When I started to write the current UHK firmware, I based it off of the [usb_device_composite_hid_mouse_hid_keyboard demo](https://github.com/UltimateHackingKeyboard/KSDK_2.0_MK22FN512xxx12/tree/master/boards/frdmk22f/usb_examples/usb_device_composite_hid_mouse_hid_keyboard/bm) of KSDK, removed everything I didn't need, and added what I needed. The remnant of this demo is the [ksdk_usb](https://github.com/UltimateHackingKeyboard/firmware/tree/master/right/src/ksdk_usb) directory of the firmware. Given this background, all of the current UHK interfaces are HID interfaces and ksdk_usb might have to be extended or modified to deal with non-HID interfaces. It might be worth to check out other [KSDK USB examples](https://github.com/UltimateHackingKeyboard/KSDK_2.0_MK22FN512xxx12/tree/master/boards/frdmk22f/usb_examples) regarding how the KSDK deals with non-HID interfaces.

As for the VID/PID issue, I'm very surprised about the Linux implementation. I guess we can just use one of these VID/PIDs for development purposes and later we can get ours added to the list. I don't have a clue about Windows (and OSX) either. We can figure it out later, but again, for development purposes we can just use an already existing VID/PID.

I think it'd be a good strategy for you to focus on the USB and firmware part of this now. You could simply temporarily repurpose a very rarely used set of keys for Xbox gamepad buttons, for example F13-F24, then make the firmware ignore these keys for the keyboard interface, and use them for the
newly created Xinput interface. Then you could configure them in Agent without modifying Agent in any ways. Once you'll have the USB part working, you can count on us to make this happen in Agent.

Keep it up, and please keep me posted! I'll try to help if needed.

ata2001, 2018-02-05:
Here is my current progress: https://github.com/ata2001/firmware/commit/446eba2d4838de131a64580de1d1042e7703efb0

I'm not sure it works, because I messed up the descriptor once, and now I can't reenumerate my UHK. If I understand correctly, I should use the command 0x01 with mode 0 on EP1, though it's not that easy because of the broken descriptor. The js script can't find the device, pyUSB is giving me errors.

lsusb says:
`Couldn't get configuration descriptor 0, some information will be missing`

mondalaci, 2018-02-05:
By the looks of your changes, it seems that the implementation of the new USB interface is missing. `usb_interfaces/usb_interface_gamepad.c` is entirely commented out, and `usb_composite_device.[ch]` are also unchaged apart from the rename. I only skimmed through your code, but this is my best guess.

ata2001, 2018-02-05:
@mondalaci You're right, I didn't actually implement it yet, I just wanted to check the descriptor, but now I don't know how to get into the bootloader. Should I write two bytes, 0x01 and 0x00 to the first HID interface (EP 1 IN)?

mondalaci, 2018-02-05:
Until you fix the firmware, you can only jump to the bootloader by shorting the RESET and GND pins of the right half. There's an incomplete guide for [partially disassembling the left half](https://ultimatehackingkeyboard.com/repair/led-display). The procedure is similar for the right half. These are the relevant pins to short:

[image]

If you have a soldering iron, you can even solder a reed switch to the other side of the PCB and then you'll be able to jump to the bootloader when putting a magnet near the back of your UHK. That way you can have your UHK assembled all the time and still be able to jump to the bootloader via hardware means easily. I can show how to do it if you're interested.

ata2001, 2018-02-05:
@mondalaci Thank you for the instructions! I was finally able to flash the firmware, now I can start trying to figure out what's wrong with the descriptor.

ata2001, 2018-02-07:
@mondalaci Tell me if I'm being spammy, because this is not much progress, but here's how the lsusb report looks like now, that I fixed the descriptor:

```
Bus 006 Device 005: ID 1d50:6122 OpenMoko, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1d50 OpenMoko, Inc.
  idProduct          0x6122 
  bcdDevice            1.01
  iManufacturer           1 Ultimate Gadget Laboratories
  iProduct                2 Ultimate Hacking Keyboard
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          166
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      31
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
[... 129 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
Device Status:     0x0001
  Self Powered
```

The last interface should look similar to the xinput gamepad in the original post.

mondalaci, 2018-02-07:
You're not being spammy at all! I'm very excited to see your progress. The descriptor looks good, and it's a big step forward that the new interface properly enumerates.

What's next? I assume that you'll make the interface send reports according to keypresses. Will you temporarily repurpose a sparsely used subset of the scancodes (such as F13-F24) as suggested?

ata2001, 2018-02-07:
@mondalaci Now I'm going to define the input report struct, and implement the updater function. I think the USB_DeviceHidSend() function should work just fine for sending xinput reports.

Yes, I'm going to map a button press to an already existing key for testing purposes, I don't want to mess with the config parser right now.

Lauszus, 2018-02-14:
@ata2001 can you post the raw descriptors when the joystick is in XInput mode? Also can you post the HID descriptors? Btw have a look at the following macro for generating a generic HID joystick: https://github.com/UltimateHackingKeyboard/firmware/blob/master/right/src/lufa/HIDClassCommon.h#L354-L382.

ata2001, 2018-03-29:
@Lauszus I exported the raw bytes from wireshark, if that is what you meant.

https://github.com/ata2001/gamepad-descriptors

(This is Logitech F310 Xinput mode)

ata2001, 2018-04-01:
HID mode:

```
Bus 003 Device 003: ID 046d:c216 Logitech, Inc. Dual Action Gamepad
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc216 Dual Action Gamepad
  bcdDevice            4.14
  iManufacturer           1 Logitech
  iProduct                9 Logitech Dual Action
  iSerial                 3 XXXXXXXX
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      99
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
```

Thanks for the link by the way, will check it out.

mondalaci, 2018-06-27:
Realistically speaking, this is a very niche feature which would demand significant resources to implement both on the firmware side and Agent side. We have to deal with a lot nowadays, and I don't want to devote resources for this until we have nothing else to do. Now I'm closing it and labelling as an exotic feature for future reconsideration.

closed 2018-06-27 by mondalaci

benedekkupper, 2022-11-04:
I should have found this ticket earlier, the XInput interface is added to the current version of the firmware (but the functionality is not added yet). See #535