UltimateHackingKeyboard/firmware #99: Mouse pointer slowness

mondalaci, 2018-04-07:
A [seemingly harmless code change](https://github.com/UltimateHackingKeyboard/firmware/commit/a84b883e40d626f53235f0b252d62d1a58cb03d8) causes the mouse pointer to move very slowly when using the release build. When I put `EmptyFunction()` in `usb_command_apply_config.c`, the mouse pointer doesn't get slow. My guess is that GCC unoptimizes something when a function is referenced from a different compilation unit.

I also tested the debug build, and it seems that the mouse pointer is already slow with it, even in the master branch regardless of this commit. So the root cause of this issue is something else. Nevertheless, it'd be interesting to figure out why GCC behaves this way, but the root cause must be solved, too.

Lauszus, 2018-04-08:
I can reproduce this issue on Mac as well.

mondalaci, 2018-04-08:
The raw configuration values are multipied by 25 for mouse movement and 1 for mouse scroll events, hence the relevant `mouse_kinetic_state_t.intMultiplier` values.

commit d2acfba6 (2018-04-08): Process the key inputs at a constant rate when moving the mouse, so the mouse speed is consistent

Fixes #99
closed 2018-04-10 by mondalaci