UltimateHackingKeyboard/firmware #105: Keyboard fully bricked when flashing firmware 8.1.5 with agent 1.1.1

Jopie01, 2018-04-10:
I just wanted to flash the newest firmware onto the UHK. but I got a timeout and after that the UHK died. No lights nothing :-( 

When I insert the USB cable I got"
```
[  369.059479] usb 3-1: new full-speed USB device number 3 using ohci-pci
[  369.225696] usb 3-1: New USB device found, idVendor=1d50, idProduct=6120
[  369.225704] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  369.225708] usb 3-1: Product: UHK Bootloader
[  369.225712] usb 3-1: Manufacturer: Ultimate Gadget Laboratories
[  369.237468] hid-generic 0003:1D50:6120.0004: hiddev97,hidraw2: USB HID v1.00 Device [Ultimate Gadget Laboratories UHK Bootloader] on usb-0000:00:12.0-1/input0
```

Nothing more :-(  It seems that the UHK is now in bootloader and waiting for something? What to do now?

Jopie01, 2018-04-10:
Nope, it's not that easy :-( .... starting the Agent says "Could not find your UHK, please plug it in". So that was a dead end. But I don't give up that easy and thought by myself, how does the agent the flasing? I digged a bit deeper into the issue and found out something about blhost and flashing. Crossed my fingers and did:

1. Download the lasted firmware (1.8.5) and untar it
2. Downloaded the right blhost binary from https://github.com/UltimateHackingKeyboard/bootloader/tree/master/bin/Tools/blhost and made it executable
3. followed the instructions from the readme of https://github.com/UltimateHackingKeyboard/bootloader#uploading-the-application-firmware-via-the-bootloader
  commands were:
  - blhost --usb 0x1D50,0x6120 flash-erase-all 0
  - blhost --usb 0x1D50,0x6120 flash-image uhk-firmware-1.8.5/devices/uhk60-right/firmware.hex
  - blhost --usb 0x1D50,0x6120 reset
After some tense moments the leds on the left half came up again! YES! Succes!
The last thing I did was starting the latest agent and flashing the latest firmware again, now to both halves. So I'm typing this again from my very beloved UHK :-)

After looking into what was going wrong, it's completely my fault. I downloaded the sourcecode and flashed that to the UHK instead of the builded uhk-firmware.

Maybe add this to the documentation for people who are as stupid as I am?

Jopie01, 2018-04-10:
I was just thinking: Should the last blhost command be enough to get the UHK back? And if so, if the agent puts the UHK first in "bootloader-mode" before flashing, then you should always execute the last command `blhost --usb 0x1D50,0x6120 reset` even if flashing fails.

borje, 2018-04-10:
Couldn't UHK Agent do a sanitycheck of the file before flashing it?

//Börje

mondalaci, 2018-04-11:
@Jopie01 Excellent job on unbricking your keyboard! Now that I read your original log, it's apparent that the right keyboard half was bricked which Agent can't unbrick yet, so I've just opened https://github.com/UltimateHackingKeyboard/agent/issues/607 - I think this will be the best and most user friendly way forward as one can unbrick his/her UHK accidentally.

@borje The bootloader already maintains a checksum. That's why it can take over instead of executing a corrupted (half-transferred) firmware image. I don't think other sanity checks are worth to implement.

Closing this issue, but feel free to follow up regardless.

closed 2018-04-11 by mondalaci