UltimateHackingKeyboard/firmware #106: Fade out / fade in LEDs upon suspend / resume

mondalaci, 2018-04-11:
instead of suddently disabling / enabling them.

Lauszus, 2018-06-25:
I had a quick look at this earlier. One easy way to do it would be to use a separate frame where all LEDs are off and then use the "breath functionality": http://www.issi.com/WW/pdf/31FL3731.pdf to fade between the two frames.

I also found some code written in Python: https://github.com/adafruit/Adafruit_CircuitPython_IS31FL3731/blob/f64bc7dccbb65f591b12e15a105feb3ce144c034/adafruit_is31fl3731.py#L189-L215.

The fade effect could be set like something like this:

```C
float fade_in_ms = 1000.0f; // Range: 26 - 3328 ms
uint8_t fade_in = (uint8_t)roundf(log2f(fade_in_ms / 26.0f));
float fade_out_ms = 1000.0f; // Range: 26 - 3328 ms
uint8_t fade_out = (uint8_t)roundf(log2f(fade_out_ms / 26.0f));
float pause_ms = 1000.0f; // Range: 26 - 3328 ms
uint8_t pause = (uint8_t)roundf(log2f(pause_ms / 26.0f));
static uint8_t setFadeBuffer[] = { 0x08, fade_out << 4 | face_in, 1 << 4 |  pause };
status = I2cAsyncWrite(ledDriverAddress, setFadeBuffer, sizeof(setFadeBuffer));
```

owlstead, 2018-11-10:
Seems like a low prio feature request. Assume it is better to tag it as such - as it is still open.

mondalaci, 2019-12-13:
Unfortunately, the breadth functionality is unavailable in the ISSI LED driver ICs that we plan to use in further UHKs, making the smooth implementation of fading infeasible.

closed 2019-12-13 by mondalaci