UltimateHackingKeyboard/firmware #109: Secure firmware upgrades

mondalaci, 2018-04-30:
See https://ultimatehackingkeyboard.com/blog/2015/02/26/securing-firmware-upgrades

owlstead, 2018-11-19:
Instead of relying on security modes you could also consider signing the firmware and bootloader. However, this might not be "hackable" enough for you. It will be interesting to see this "exiting" feature appear. Generally, however, if security is added afterwards it takes a back seat to bugs and feature requests (I'm a security expert and software architect).

rolag, 2021-05-09:
Does this still apply to the V2? And is there anything someone could do to help complete this issue faster?

mondalaci, 2021-05-10:
Yes, it applies to the v2, too. I can't provide you an ETA on this as we're busy with other issues.

mondalaci, 2023-09-06:
I don't think it's worth implementing this feature since nobody has requested it, and no security incidents were reported. If we ever implement something like this, there'll be a simple hardcoded policy of keeping certain key(s) pressed for a while. For the time being, I'm closing it.

Ofenhed, 2023-09-06:
That's disappointing... maybe I'm the only one (as I'm the Marcus referred to in the blog post), but the lack of security is the one and only reason why I sold my UHK a few years back, even though I absolutely loved using it. I feel that your target audience with an expensive "hacking keyboard" should have an unusually high percentage of security aware people. If you're not hearing about it, is it because it really isn't an issue people care about, or have security aware people just removed UHK from consideration?

mondalaci, 2023-09-06:
The 4 corner keys are reasonable. Thanks for your opinion!

josephvusich, 2023-09-07:
> The lack of upvotes on this issue suggests to me that people don't really care.

Speaking personally, I subscribed to this issue without upvoting. I would really like to see this feature implemented.

Currently I've mitigated the security concerns by putting the UHK behind a KVM switch that only reports its own virtual HID to the OS. The Agent does not even see the UHK when connected in this manner.

When a firmware upgrade or configuration change is required, I plug the keyboard in directly, bypassing the KVM. This is my workaround for the lack of hardware security on the UHK. I would very much prefer the convenience of a write-protection feature on the UHK itself, as this would eliminate the need to physically swap connections.

adrianimboden, 2024-01-23:
I did not work on the keyboard software myself yet, but I would be interested to help out when you point me to the relevant parts. The security part always comes up after I change the configuration and it bothers me from then on for some weeks until I forget about it.

When detecting a key (or multiple keys) at bootloader stage is easy (e.g. you have a code sample where we do that already in the bootloader), I will happily implement and test it out.

mondalaci, 2024-01-23:
Thanks for offering your help, and I don't want to discourage you, but our experience is that helping occasional contributors takes us more time than doing the task ourselves, so we'd rather implement this feature.