UltimateHackingKeyboard/firmware #117: Mouse Clicking Issue

sad2project, 2018-05-24:
To start with, I will point out that I changed quite a bit around from the defaults. My Mouse key is the left chassis switch, and I moved the arrow keys to the left side (E,S,D,N) and the clicking and scrolling to the right (J,<,L are the clicks with scrolling being I,F,U,O). There are many more changes, and I could upload my config if you like.

The problem is that sometimes (usually, if not always after doing a lot of mousing and clicking around without letting go on the Mouse button) my left click gets messed up. In the browser, clicking on a link acts like I right clicked and selected "Save link as...". In windows explorer, clicking (possibly only double clicking) on a file opens the file's properties.

Letting go of the Mouse button then trying again usually works, if not the first time, but if I never let go of the Mouse button, it seems to never stop.

mondalaci, 2018-05-24:
Hi Jacob,

Please export your configuration and attach it. I'd like to reproduce this
issue to fix it.

Thanks for your report and patience!

sad2project, 2018-05-25:
[UserConfiguration.zip](https://github.com/UltimateHackingKeyboard/firmware/files/2037543/UserConfiguration.zip)

I'll be testing it with my other keyboard and computer to see how localized this issue is.

sad2project, 2018-05-25:
[UserConfiguration.zip](https://github.com/UltimateHackingKeyboard/firmware/files/2037553/UserConfiguration.zip)

It should also be noted that this is only confirmed on 8.2.2. I'll let you know if it continues with the new 8.2.4 firmware update that I just noticed existed.

mondalaci, 2018-05-25:
I tried your config with the 8.2.2 firmware and couldn't reproduce the behavior. I don't think it's related to your configuration because the mappings should behave consistently. I also don't think that this was resolved in more recent firmware versions. The fact that the left and right mouse buttons get accidentally swapped is very weird.

1. Do you use any device between your UHK and the host computer like a USB hub or a KVM switch? If so, please remove them, and connect your UHK directly to the host computer.
2. Please make sure to use your UHK with a host OS, not a guest (VirtualBox, VMware Workstation).
3. Try to remove every other USB devices.
4. Try a different host computer.

Please give the above a try, and let me know your findings.

sad2project, 2018-05-31:
I can verify that the recent update didn't fix it. Should finally have the chance to do the controlled test properly tonight.

mondalaci, 2018-05-31:
Thanks, I'm waiting for your findings!

sad2project, 2018-06-01:
Was able to reproduce with my other UHK on a different computer.

You have to pretty much hold the mouse button for several minutes straight, with jumping around on the browser being one of the easiest ways to do it.

Also, I want to make it clear, this doesn't cause the "buttons to be swapped" like you said earlier. It chooses an item from the right click menu. I have no clue how it determines which item, but it's consistent. "Save as..." in Chrome and "Properties" in Windows explorer. Haven't experienced it anywhere else, so I don't know what it would choose in other situations.

sad2project, 2018-06-01:
Ah, didn't mean to close it!

mondalaci, 2018-06-01:
This is a very hard one to reproduce. I've held the mouse key for 10 minutes and still couldn't reproduce it. I could really use some more help.

The more specific, exact, and easier instructions you can provide, the more likely I can reproduce this issue. Please narrow it down as much as possible by looking into the following:

Can you reproduce this issue:
1. on the default keymap of the default configuration (after you reset your keyboard configuration)?
2. after x minutes of holding the mouse key?
3. after x {right/left/middle} clicks?
4. not only by holding the mouse but by using the mouse layer in a toggled state?
5. without using scrolling?

I know it's hard, and this issue may not be deterministic, but narrowing it down as much as possible would help tremendously.

Thanks for everything!

Hurricaaane, 2018-06-10:
> In the browser, clicking on a link acts like I right clicked and selected "Save link as...". In windows explorer, clicking (possibly only double clicking) on a file opens the file's properties.

This is ALT-clicking.
- ALT-double click on file explorer = Open properties
- ALT-click on a link = Save link as

sad2project, 2018-06-11:
confirmed

luteijn, 2018-06-18:
Seems that the same thing happens for **ctrl**, turning mouse scrolling into zooming until one unsticks the ctrl key. For example, switch to mouse layer with a double tap, scroll with 'y' / 'h'. tap control once, 'y' and 'h' now zoom. Fixed by leaving mouse layer.

mondalaci, 2018-06-18:
@luteijn Thanks, I can clearly reproduce your issue.

luteijn, 2018-06-18:
Looking at this with 'xev', the same thing seems to happen in the **mod** and **fn** layer, probably just less obvious effects there, but **shift**, **ctrl**, **super**, **alt** all get stuck down when used until layer is left again. This may make things easier to fix as it seems to be a general issue, not just the **alt** modifier, but _all_ the modifiers on _all_ non-base layers. On the other hand, there may be a secret reason for this behavior that might mean fixing it breaks something else.

mondalaci, 2018-06-18:
@luteijn Thanks, I'm aware of the above.

luteijn, 2018-06-18:
Looking through the source, I think this is indeed caused by a 'feature':
    // When a layer switcher key gets pressed along with another key that produces some modifiers 
    // and the accomanying key gets released then keep the related modifiers active a long as the 
    // layer switcher key stays pressed.  Useful for Alt+Tab keymappings and the like.
So probably need to decide when holding is the right thing to do and when not, and when to release a held modifier key again and include this logic in the firmware. Maybe start simple with a global choice for the user if they want to hold modifiers like this or not (I'd prefer not, as it also doesn't get done for the base layer, and I'd like all layers to be equal), or just don't hold when layer is locked instead of just temporarily activated by a held key. 
This may get fixed/be worse when those modifier keys also turn into layer switchers, to be able to send different scan codes for a key when 'modified'.

TorC8, 2018-07-04:
I just checked the behavior of the Alt+Tab window switch, and I agree the two points above should fix the issue.  On my system, M-Tab and M-S-Tab cycle forward and backward, respectively.  Holding shift while releasing Alt still maintains the window list, allowing an Esc to be used to cancel the window switch.

However, there is the issue when using, for example, the default Mod-D window switch.  Esc is used to cancel, but by dropping the modifier before sending the Esc the expected behaviour of the Esc key is subverted.

I'm wondering if the correct behavior is to special case Esc here.  If there are sticky modifiers from a modifier+scancode key, send Esc before dropping the modifiers.  In all other cases drop modifiers not set by a held key before sending the new scancode.

mondalaci, 2018-07-04:
@TorC8 I understand that you're concerned about this use case, but I also believe that it's very rare and I'm afraid that introducing a special case regarding Esc would hurt consistency, and might be problematic in other cases. I wouldn't make an exception in this case.

mhantsch, 2018-07-13:
@TorC8 what would happen if you configured the Esc to send Alt-Esc on that layer? I would hope that when the Alt modifier is held by UHK, it would keep it down (instead of releasing and immediately pressing it again for the Alt-Esc). In summary, it should send Alt-Tab, keep Alt held, then send Alt-Esc, and finally release Alt.

So, would your use case be fixed by just reconfiguring Esc to Alt-Esc (on that layer)?

luteijn, 2018-07-14:
Basically as I understand things, all this kludgery is to allow people to not use alt-tab and alt-shift-tab to go through their list of windows, but another modifier key (to switch layers) instead of alt, and another key that is not tab. Is it really worth it to cause things not to work as expected in all the other use cases, that (at least to me) seem more important than window manager control? The 'hold implicit modifier to facilitate keeping the window list open' functionality seems to be the 'exception' here. Probably better of by just making 'double tap to lock', 'single tap to lock', 'do not lock' an option to set for every key (even if most people would just want layer switching / modifier keys to have that function).

mondalaci, 2018-07-15:
@luteijn Your understanding is correct, but this should be rendered to a non-issue once the fix is out.

mhantsch, 2018-07-30:
@luteijn Just for clarification why some users want such "kludgery", here is one of my use cases: I often end up pressing Left Fn instead of Left Alt with my thumb because regular keyboards do not have a Fn key and I guess my thumb slides from the Space bar towards the left until it subconsciously finds the next key. So I want to press Alt-Tab and end up pressing Fn-Tab instead. That's fine, I can configure Tab to behave like Alt-Tab on the Fn layer, and tabbing is good. Also I have now changed Fn-\` to send Alt-Esc and not only Esc so I can abort tabbing while holding Fn, so that use case works for me.

The one configuration where this does not work is if the layer switching action is only the secondary action on the modifier key. See https://github.com/UltimateHackingKeyboard/firmware/issues/125

luteijn, 2018-07-30:
Perhaps just swapping alt and fn keys would be the better solution then? Then the alt key would be where you expect it to be again. One might also want to put super one to the left. That's what I did at least.

PR #149 (2018-08-02): Fix various secondary role bugs
This closes #145, closes #125, closes #119, closes #117, and closes #69.
closed 2018-08-02 by mondalaci