UltimateHackingKeyboard/firmware #118: Firmware update doesn't work via USB3

mondalaci, 2018-05-28:
Only via USB2 ports.

eltang, 2018-05-28:
I've had no problem using the USB3 ports on my Mac.

mondalaci, 2018-05-28:
@eltang Thanks for the feedback! What kind of Mac and OSX version exactly?

eltang, 2018-05-29:
I have a MacBook Pro running macOS 10.13.4.

kour1er, 2018-05-29:
Just updated to Agent 1.2.2 with firmware 8.2.5 and have successfully flashed from a USB3 Mac Pro (6,1 A1481) with Host Controller Driver: AppleUSBXHCIFL1100. It works with both a direct connect and via a power USB3 hub. This is my first non-USB2 successful flash. So whatever changed recently, fixed it for me. 👍

kour1er, 2018-06-04:
Sorry for taking a while to get back on this one. I have conducted the following tests to try and track this down. The test machine is a Mac Pro (6,1) running macOS 10.13.5 and my UHK is connected via a powered USB3 hub.

I first wanted to test the firmware flashes via their respective versions of Agent with which they shipped. I found that most of the older versions of Agent no longer run:

1.0.0 Agent doesn't recognise a keyboard anymore on 10.13.5
1.0.1 "cannot talk to your UHK" (invalid platform darwin)
1.0.2 "cannot talk to your UHK" (invalid platform darwin)
1.0.3 "cannot talk to your UHK" (invalid platform darwin)
1.0.4 "cannot talk to your UHK" (invalid platform darwin)
1.1.0 "cannot talk to your UHK" (invalid platform darwin)
1.1.1 "cannot talk to your UHK" (invalid platform darwin)
1.1.2 "cannot talk to your UHK" (invalid platform darwin)
1.1.3 "cannot talk to your UHK" (invalid platform darwin)
1.1.4 "cannot talk to your UHK" (invalid platform darwin)
1.1.5 "cannot talk to your UHK" (invalid platform darwin)
1.2.0 "cannot talk to your UHK" (invalid platform darwin)
1.2.1 runs (flashed with 8.2.2 ok)
1.2.2 runs (flashed with 8.2.5 ok)

I don't know if this relates to the underlying Electron technology in relation to my current version of macOS or something else, but thought this information might be useful.

Next I thought that trying to flash the various versions of the firmware with the most recent version of Agent might provide useful information. In these tests, my UHK was reset to 8.2.5 in-between every flash to provide a consistent testing experience:

7.0.0 flashes (FTY requires unplug replug) but left half doesn't work
8.0.0 flashes (FTY requires unplug/replug) but left half doesn't flash
8.0.1 flashes (FTY requires unplug/replug) but left half doesn't flash
8.1.1 flashes both halves (FTY requires unplug/replug). Invalid hardware configuration error
8.1.2 flashes both halves (FTY requires unplug/replug). Invalid hardware configuration error
8.1.4 flashes both halves (FTY requires unplug/replug). Invalid hardware configuration error
8.1.5 flashes both halves (FTY requires unplug/replug). Invalid hardware configuration error
8.2.0 flashed both halves and no FTY unplug/replug
8.2.1 flashed both halves and no FTY unplug/replug
8.2.2 flashed both halves and no FTY unplug/replug
8.2.5 flashed both halves and no FTY unplug/replug

Hope this provides some useful information.

mondalaci, 2018-06-04:
@kour1er Thanks for your feedback! Very much appreciated!

The failed firmware update attempts of your first set of tests were due to the "Match for the new USB usage page and usage number. This is critical for UHKs flashed with firmware >=8.2.2 to be recognized by Agent on OSX." changelog entry of [Agent 1.2.1](https://github.com/UltimateHackingKeyboard/agent/releases/tag/v1.2.1).

As for your second set of tests, firmware [8.2.5](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v8.2.5) solved the FTY issue completely. Your test with earlier firmware versions were successful due to the nondeterministic nature of this bug.

Given the above, I'm confident that the " "Fix vendor-specific USB usage page entry." fix of firmware 8.2.2 fixed this issue, and I consider this issue to be resolved.

closed 2018-06-04 by mondalaci