UltimateHackingKeyboard/firmware #122: Keys got stuck after about an hour of firmware startup

mondalaci, 2018-06-07:
Updated to firmware 8.3.1 and after about an hour of use following firmware startup, the actual key gets virtually stuck. I can't get it unstuck, only by power cycling the keyboard. Today, it happened to the "e" key, then Alt+Tab (via Mod+d), and finally the down arrow (via Mod+k).

This is by far the most critical bug as we can't make an upcoming release until it gets fixed.

mondalaci, 2018-06-07:
Now it happened in 10 minutes. The left shift got stuck, and keyboard input got totally unresponsive.

mondalaci, 2018-06-10:
@Lauszus In the meantime, the bug happened about 10 times on my keyboard. The current fix doesn't seem to improve the situation at all. Any ideas how to solve it?

Hurricaaane, 2018-06-10:
Not sure if this is related, but my keyboard acted strangely as if it was partially responding, a couple hours after I installed the latest released firmware. I've downgraded it for now.

mondalaci, 2018-06-10:
@Hurricaaane Yes, pretty sure this one is the culprit. I've just marked the latest firmware as a pre-release instead of a release, and added a comment regarding this bug in the release. Hopefully, this will prevent others of installing the latest release and hitting this bug.

vvolkgang, 2018-06-11:
Installed due to the media repetition on windows fix but, this happened twice, had to power cycle it.

Also, after some variable time, UHK stops sending keystrokes to the OS, don't know if it's related to this specific issue but only happened after flashing 8.3.1. Laptop keyboard works, mod / fn / mouse status work too. Power cycling the UHK fixes the problem.

mondalaci, 2018-06-11:
@vvolkgang Yep, sounds like this issue. Please downgrade to firmware 8.3.0 until we resolve this.

laxu, 2018-06-16:
I've had both happen on my UHK. Keystrokes keep repeating or just stop getting sent. Disconnecting and reconnecting the USB cable fixes this. Hopefully this gets solved soon because I love that I have working sleep and media keys again.

mondalaci, 2018-06-16:
@Lauszus This happened again and this time I captured some packets with Wireshark: [packets.zip](https://github.com/UltimateHackingKeyboard/firmware/files/2108303/packets.zip)

* uhk-idle.pcapng: didn't touch any keys
* uhk-typing.pcapng: hammered the keys quickly
* uhk-mouse-media.pcapng: navigated around with the mouse, then used media keys

Lauszus, 2018-06-20:
@mondalaci I had a look at the logs. Sorry but you need to start capturing the data before the bug happens or I won't be able to see what is sent just before it occurs.

mondalaci, 2018-06-22:
@Lauszus Here's the capture: [keyboard-interface-dies.pcapng.gz](https://github.com/UltimateHackingKeyboard/firmware/files/2127575/keyboard-interface-dies.pcapng.gz)

vvolkgang, 2018-06-22:
Don't know if it's relevant to this issue but the previous firmware is
locking MOD at least, can't find the pattern, usually happens when I'm
clicking other modifiers and doing several stuff with the mod key pressed
without releasing it (e.g. select several words with mod ctrl and shift,
delete, go up, select more, delete, everything without releasing the mod
key).

Then, mod won't turn off even if I double tap it. Per usual, power cycle
fixes it.

mondalaci, 2018-06-22:
@vvolkgang #121 is unrelated and has already been fixed.

Lauszus, 2018-06-22:
@mondalaci thanks. I will take a look at it.

PR #123 (2018-07-01): Fix issue with keyboard freezing
* Fixes #122 
* Fix PIT timer period and added functions for getting the current time in microseconds - see #127
commit 9be3cf48 (2018-07-01): Invalidate the stored report if the command fails

Fixes #122
closed 2018-07-01 by mondalaci