UltimateHackingKeyboard/firmware #124: Alt seems to cancel Mod layer

irwand, 2018-06-14:
Not sure if this is fixed or if this is user error. I'm on Agent 1.2.2, with firmware 8.2.5.
I can't figure out how to press Alt-F4. To do this, I'd press Alt+Mod+4 key, but it doesn't work, because I think it registers as Alt-4, instead of the intended Alt-F4.
This doesn't happen with Shift key. The Shift key act as intended.

irwand, 2018-06-14:
Oh and this also doesn't happen with Win key (super). I can press Win-Mod-1, and it registers as Win-F1 correctly. Seems like this only happens with Alt key.

mondalaci, 2018-06-14:
Mod+Alt+4 should definitely trigger Alt+F4. Please check in Agent how your keyboard is mapped. It's quite likely that the Alt key is not the one you think it is. Please get back to me and let me know your findings.

irwand, 2018-06-14:
Oh!
OK, so what I did was that I swapped Fn and Alt key. But then the Alt key isn't assigned anything when the Mod layer is on. I had thought that those ctrl-shift-alt would be unmodified when different layer is active, which is invalid assumption. Sorry for the user-error.

closed 2018-06-14 by irwand

mondalaci, 2018-06-14:
Thanks for getting back to me so quickly! Yes, this might not be obvious. This is by design, so that modifiers are configurable on a per layer basis. Upcoming Agent versions will feature a "remap on all layers" checkbox to address this use case.

irwand, 2018-06-14:
Oh, that'd be awesome! Thank you much for you and your team's hard work! This keyboard ROCKS! I love it!