UltimateHackingKeyboard/firmware #125: Secondary-role layer switch causes weird alt-tab behaviour

mhantsch, 2018-06-18:
I assigned the left space to send Space scancode and secondary role to switch to the Mouse layer.
On the Mouse layer, I configured Tab to send LAlt-Tab.
When I press Space-Tab, the windows switch immediately (to the previously active window). If I keep holding down the Space key, I cannot keep tabbing through all the open windows.

When instead I use a primary role switch to the Mouse layer, everything works as intended, and I can tab through the open windows while keeping the layer switch key depressed. However, doing the same with a secondary role switch, UHK only seems to send a single LAlt-Tab and immediately releases LAlt although I keep that secondary role key depressed. I think it should keep the LAlt modifier depressed until I let go of the secondary role key.

mhantsch, 2018-06-18:
This might be related to #69

mondalaci, 2018-06-18:
Thanks for reporting! I can clearly reproduce this issue.

TorC8, 2018-06-18:
That opens up the question of right behavior.  Certainly I think it should hold for repeated presses of the key, but what about the case of, while still holding $MOUSE, pressing a different key that doesn't have $ALT configured as part of its keycode?

My inclination is to think that the right thing is to drop the modifier at that point before sending the new keycode, but I can see arguments for both options.

mondalaci, 2018-06-19:
I agree that dropping Alt is the right thing to do in the above case. I can't see why any other option would make sense.

mhantsch, 2018-06-19:
Unless it's Shift; then you need to keep Alt held down because the user is trying to type Alt-Shift-Tab to cycle the windows backwards. So, I would argue: drop Alt only if the next keypress produces an actual character, macro, layer switch, or keymap switch; don't drop it on modifiers-only (Ctrl/Shift/Alt).

mondalaci, 2018-06-19:
@mhantsch Yep, we're on the same page.

closed 2018-08-02 by mondalaci

mhantsch, 2018-08-10:
I can confirm that this is resolved with firmware 8.4.2.

PR #149 (2018-08-02): Fix various secondary role bugs
This closes #145, closes #125, closes #119, closes #117, and closes #69.