UltimateHackingKeyboard/firmware #128: Occasional double input

linduxed, 2018-06-24:
I'm experiencing frequent but seemingly irregular occurrences of double input from pressing certain keys. This causes misspellings such as "dotted" becoming "dottetd".

The key that seems to be doing this the most is (on QWERTY) the `T` key. Other keys that exhibit this behaviour are `I`, `S`, `F`, and `Space` (right half, next to `Fn`).

There are probably other keys that I've witnessed producing a double input, but I can't remember them off the top of my head.

If one was to focus on the most frequently misbehaving key: I'm guessing that my habits of how I press keys, combined with where `T` is positioned triggers some kind of edge case.

mondalaci, 2018-06-24:
@linduxed Thank you for reporting this issue! I've also experienced some chatter lately, and we received about 2 reports so far from our backers. I'll ping the reporters to share their experiences here so that it'll be easier for me to fix it. I have some ideas how to track this down.

linduxed, 2018-06-24:
Updated to `8.2.5`. Issue still present.

kour1er, 2018-06-26:
@linduxed out of curiosity, what OS are you using? I'm wondering if there are difference between OS and UHK with this. macOS here and seeing the issue

linduxed, 2018-06-26:
@kour1er Arch Linux.

kour1er, 2018-06-26:
Interesting - so not limited to a specific OS. The plot thickens :)

kareltucek, 2018-06-26:
@kour1er  This clearly happens in case of bad connection of switches (either inside of the switch or somewhere around)(in my case this happens with loosely connected unsoldered switches)*, yet there may probably be some other hardware-or-firmware-design-related issues.

I am quite curious if there is some relation to switch type/brand. 

Simple testing procedure - get a key which produces a key as primay action and has some secondary action. Then hold, change pressure, push the key into sides etc.. Every skip will produce aa keystroke...

*(I am of course not complaining about that :-) .)

kour1er, 2018-06-26:
@kareltucek interesting. So you think that this is a fundamental hardware issue? It's so weird because when I first got my UHK I had zero chatter issues. It's only over the last month and a half that I've been seeing it.

kareltucek, 2018-06-26:
Well, from my obervations, I am confident that one possible cause is this hardware one - minor malfunctions of the mechanical switches which are detected by hypersensitive firmware. I do not believe that the problem is between firmware and PC. I believe that it is in the way firmware interacts with switches. I would not call it fundamental hardware issue - I would rather say that firmware's assumptions about hardware's behaviour are a tiny bit off. I believe that it will be possible to fix in firmware.

However, this is my personal opinion based on my observations. Moreover, these obervations are based on having done quite barbarous modification to my hardware and therefore may be *totally* irrelevant. I am sure Laci will be able to provide much more relevant explanation.

kour1er, 2018-06-26:
@kareltucek what a great way of explaining it - thank you

And I am now so curious about your barbarous modifications!

linduxed, 2018-06-26:
> @kour1er It's so weird because when I first got my UHK I had zero chatter issues.

Strange, I also recall this being a somewhat new issue for me. I think that it's only been going on for the past two weeks or so. IIRC I got my UHK a month and a half ago.

mondalaci, 2018-06-26:
The characteristics of the of switches may change over time, but the firmware should deal with such issues. I plan to get to the bottom of this, but I have to deal with a lot nowadays. I'll follow up.

kour1er, 2018-06-26:
@mondalaci I can't imagine how many things you're juggling at the moment! Thanks for following up

mondalaci, 2018-06-27:
Our intern, @eltang has taken a quick look at the firmware debouncer which was followed by this conversation:

> Eric: So I think the issue might be that debouncing is only happening on press, and not release. This seems to be what's happening to the issue reporter, so if you agree I can go ahead and make a fix.
> Me: Why should debouncing happen on release?
> Eric: Because bouncing can happen on release as well. Did you not know that?
> Me: Nope :|

My cluelessness is truly limitless!

Then he went ahead to fix the debouncer accordingly. He also changed the debounce interval from 100ms to the suggested 5ms of MX switches. I was gradually increasing this value from release to release in the hope to combat this issue without realizing the root cause.

Please give a try to the newly released [firmware 8.3.2](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v8.3.2) and let us know how it works for you.

linduxed, 2018-06-27:
I just installed `8.3.2` and the issue has worsened significantly. Not only is doubling of characters now more common on the keys where I previously experienced it, on `T` (my most problematic key) now triple input now happening fairly often.

Additionally, no input from the key (which I was previously unsure of whether it's related) is also more common.

I've been trying to press the problematic keys for a while now to figure out a pattern (slow press, fast, hard, shallow, deep, fast repetitions, single stroke, etc.) and I can't figure something out.

mondalaci, 2018-06-27:
@linduxed Sorry to hear, and thanks for the quick feedback! Just uploaded [a number of firmware images](https://github.com/UltimateHackingKeyboard/firmware/releases/tag/v8.3.2) featuring different debouncing timeouts. Would you please give them a shot?

linduxed, 2018-06-27:
After trying 10ms, 50ms and 100ms, my conclusion is that 100ms is the one with the least chatter.

kour1er, 2018-06-27:
Well just tried to flash one of the new images and the flash failed and the LED on the left half of the UHK no longer lights up. Bricked? Do I need to take it apart and do the manual reset thing again? (Right side is still functional)

mondalaci, 2018-06-27:
@kour1er Just try to reflash a firmware normally and Agent should instruct you to reconnect the left half possibly multiple times to unbrick it.

kour1er, 2018-06-27:
@mondalaci Ah - you weren't kidding about multiple times! Back to life now. 

So I would agree with @linduxed that the 100ms has the least chatter. The chatter is still there, but the 100ms is much better than the 10ms version.

Out of curiosity, is the same debounce timeout applied to both the press and release keystroke?

kour1er, 2018-06-27:
One odd thing that I'm seeing with the 100ms version is some missed keystrokes. I'm seeing it mostly on the f key. Very odd.

linduxed, 2018-06-27:
Another thing about the 100ms version is that while it has the least chatter, it's enough time that it makes fast single key repetition slow down noticeably. I'll still keep this firmware until some other/better solution comes along, but doing something like fast index+middle finger alternation on the Backspace key is definitely slowed down by 100ms.

mondalaci, 2018-06-27:
Looks like we don't know what's really going on exactly. I'll expose the state of the key debouncer via USB so that you'll be able to log it, and based on the logs we should be able to fix this. I need some more time.

mondalaci, 2018-07-01:
We've just closed some major firmware issues, and this is the next on my radar. There's nothing you can do until I implement the logging feature. I'll follow up soon, I promise.

kour1er, 2018-07-01:
I don't know if you've seen this but I ran across this tool that might be helpful for testing chatter. Admittedly it's windows but that's what dual booting is for :)

[SwitchHitter](https://elitekeyboards.com/switchhitter.php)

TorC8, 2018-07-03:
I too have been having this issue (and been in contact for some time by email about it).

My experience suggests it is noticeably related to humidity.  My office has open windows just about all the time, with substantial open area.  My (possibly subjective) sense is that chatter happens more often when it's raining or recently rained than when there is a dry spell.

I live in a high humidity area, and near enough to the ocean that corrosion is a big issue here.  Is there anyone experiencing this issue who lives in a dry area?

Jopie01, 2018-07-25:
Flashed yesterday right away and so far no problems. Everything works like expected!
I had a lot of trouble entering passwords :smile:  The firmware becomes more and more mature.

luteijn, 2018-07-25:
Same here, Nice to have the macro functionality working too.

kour1er, 2018-07-25:
Just some quick feedback - not done extensive testing yet but the new firmware is a significant improvement over the old. Great work. Looking into doing the custom build to experiment with the debounce settings. 

What turned out to be the solution? Or is it a completely different approach to debouncing?

mondalaci, 2018-07-25:
Glad to hear! It's a totally different algorithm.

linduxed, 2018-07-25:
For reasons that I have not managed to figure out yet, Agent fails to build for me (something related to the building of `node-hid`).

Is `get-variable.js debounceTimePress` and `set-variable.js debounceTimePress` (and the same for `debounceTimeRelease`) something that will potentially be included as a configurable in Agent?

mondalaci, 2018-07-25:
Feel free to open an Agent issue regarding the build problem and we'll help you.

Yes, `debounceTimePress` and `debounceTimeRelease` is the same in the get and set scripts.

I don't plan to make these values configurable via Agent. Please note that they also get defaulted upon power cycling your keyboard. The idea is that we'll test this, come up with the best values (the lowest values that don't cause bouncing) and use them as factory defaults. There's no reason to mess with these values once they're right.

kour1er, 2018-07-26:
@linduxed I had to force node 8 rather than node 10 and then Agent built fine.

mondalaci, 2018-07-29:
@kour1er What is your exact methodology for tuning these values? 

The original 50-50 values did not produce bouncing on my UHK, and now I'm experimenting with lower values. A binary search algorithm is the most efficient when experimenting with the values and one should branch whether or not bouncing is experienced.

kour1er, 2018-07-29:
@mondalaci So I'm using the SwitchHitter application I referenced earlier in this thread and the macOS Debounce utility [Debounce](https://github.com/toothbrush/debounce-mac) to establish a base line. 

I started at 10 for both settings then slowly worked my way up on both values until the tools said no bounce. This happened at around the 60/50 mark. 

I then typed a lot (scientific I know, right?) and was still seeing some bouncing. Went to 70/50 then finally 70/60 and then typed a bunch more. 

The typing consisted of [TypeRacer](http://typeracer.com) and copy typing at speed some general new articles to get a variety of text (Washington Post / NYTimes type things).

Have you got a more systematic methodology I could try?

mondalaci, 2018-07-29:
@kour1er Thanks for explaining! I think you methodology is just fine. :) Did you try to change the release timeout, too? Both values should be minimal.

kour1er, 2018-07-29:
@mondalaci phew - I thought I was missing some well know methodology of keyboard testing :)

I was varying the debounceTimePress and debounceTimeRelease variables. Do you have any advice to offer on testing the two? i.e. do you start with both at say 10 and slowly increment only Press and leave Release? Or do you move both up in tandem? Or some variation?

mondalaci, 2018-07-29:
@kour1er. This is an excellent question. So excellent I'm not sure about the answer. Maybe it'd be worth do try to decrease the release timeout once the press timeout seems right.

linduxed, 2018-07-30:
@kour1er Indeed, installing `nvm` and installing Node 8 through it got it all working!

After some experimentation I'm running `25` for `Press` and `50` for `Release`. The inverse could maybe also work, although I'm not entirely sure, haven't tested it for long enough. It's also a bit tricky to figure what `Press`/`Release` does to the feeling of typing, assuming one just inverts the values.

It seems like this is something that demands a fair amount of testing.

mondalaci, 2018-08-03:
Previously, I used 10-10 (press-release) which produced occasional chatter on my side. I've been running 15-15 for several days and I can't experience any chatter.  I'm thinking about reducing release but according to your numbers guys mine may already be too low. The numbers should be high enough, so that no UHK exhibits any chatter.

@kour1er Your UHK seems to be the most problematic when it comes to chatter, and I'd be glad if you could reduce these numbers, be it press timeout, release timeout, or both.

Interested about your findings, guys.

kour1er, 2018-08-04:
@mondalaci just tried it with your 15-15 settings and seeing a lot of chatter. Out of curiosity, what's your typing speed? I'm wondering if that has an impact?

kour1er, 2018-08-04:
From my non-keyboard software writing/manufacturing perspective, I think it would be helpful to establish some sort of shared testing framework, rather than a more ad-hoc 'typing for a couple of days' and seeing how it feels.

In order to have some sort of methodology, I think it would be helpful to have a more objective measuring tool and an agreed upon set of things we are measuring. Thus, here is a list of variables that I can think of to test:

**Key characteristics:** 
- A key may be pressed soft, or hard
- A key may be pressed quickly, or slowly
- A key has a debounceTimePress variable
- A key has a debounceTimeRelease variable

**When typing a key may be pressed:**
- in isolation 
- in conjunction with an adjacent key on the same row
- in conjunction with an adjacent key on a different row
- in conjunction with a far key on the same row
- in conjunction with a far key on a different row
- in conjunction with an adjacent key on the same row on the opposite keyboard half
- in conjunction with an adjacent key on a different row on the opposite keyboard half
- in conjunction with a far key on the same row on the opposite keyboard half
- in conjunction with a far key on a different row on the opposite keyboard half

Even though it's a windows tool, SwitchHitter seems to be the best tool available (very happy if someone else is aware of a better tool?). 

**Suggested testing procedure:**

- Set SwitchHitter threshold to 10ms
- Iterate through the way a key may be pressed list
- Change one of the key characteristics and then iterate through the way a key may bee pressed list again
- Observe difference and report findings.

Does this seem like a reasonable approach? Is there a better more established methodology?

laxu, 2018-08-06:
I haven't experienced any chatter on my UHK with Cherry MX Clears. It's possible it's an issue with the switches themselves as on another keyboard I had some chatter issues that were fixed by simply replacing the switch with another one. Like you I tried adjusting debounce times but that only helped to a degree and going too high will result in the occasional dropped input.

kour1er, 2018-08-06:
@laxu Interesting. Red switches here. I would agree with you if it were the same switches that were producing chatter all the time but it's inconsistent. Leading me, from my uninformed position, to think that it might still be a firmware issue?

kour1er, 2018-08-06:
Inconsistent really is the word isn't it? It almost feels like each key press is hitting a different part of the 'key press / key released' code every time.

Even with the suggestion I made above about a more systematic testing procedure, I'm getting just different results all the time.

kbranch, 2018-08-06:
I'm wondering if there might still be some kind of firmware issue involved.  If ~5ms is the usual value for cherry switches, it seems suspicious that values an order of magnitude higher are required to (mostly) eliminate issues.

I think I'll try to take a logic analyzer to some switches tonight and see what they're actually doing compared to what shows up on screen.

kour1er, 2018-08-06:
@kbranch very interested in your findings!

Jopie01, 2018-08-06:
It can be a race-condition somewhere in the firmware or buffer overflows which also freezes the keyboard. For me using "SHIFT" or "MOD" together with a key gives sometimes chatter, but mostly "normal" keys on the right half.
It can maybe also have something to do with the USB-reports when the USB-bus is busy or the system itself. When upgrading my GNU/Linux system I often see more chatter, because the system is busy upgrading packages.

mondalaci, 2018-08-06:
I think the new debouncing code is pretty solid and certainly much better than the old code. According to my experience, there's a clear correlation between the debouncing intervalls and chatter. I'd think that the chatter is produced by a variety of physical factors such as humidity, dust, temperature, and the manufacturing conditions of the switches. These factors and the wear of the switches may change over time which can affect chatter.

@linduxed I'm surprised about the lack of registration when 50-50 is set. Haven't experienced this, and don't have an idea yet why it may happen.

@kour1er I'm not sure what else I can add. Can you experience the chatter on multiple OSes / machines? Shouldn't be OS dependent but I'm wondering.

@kbranch Logic analyzer measurements would be much welcome.

kour1er, 2018-08-07:
So in an effort to provide more data, I typed the same paragraph with my UHK in a variety of different computers. In all cases, my UHK was set to 15-15 for press release as @mondalaci is using. As you can see from the test runs, I'm still experiencing quite a bit of chatter.

The test computers:
Mac Pro running 10.13.6
Mac Mini running 10.13.6
Asus running Windows 7

I detail the connection method in the top of each test run. I include a control keyboard sample at the top (Kinesis Freestyle Pro.

[Typing Test](https://pastebin.com/LWTNSEDs)

mondalaci, 2018-08-07:
@kour1er Your UHK seems to be the most problematic of all by far. I'll get in touch with you very soon and offer a replacement unit.

kour1er, 2018-08-07:
@mondalaci Cheers and sorry to be your problem typer :)

mondalaci, 2018-08-07:
@kour1er No problem at all! You're part of the pilot run, and I'm pretty sure we messed up something.

linduxed, 2018-08-07:
@mondalaci Please note that I'm not saying the following to get a replacement keyboard:

Could my keyboard also be one that could be considered notably problematic? Should I try to do the typing test of @kour1er?

I'm mainly interested in finding some general test that people can do to determine whether they have a problematic board. I think it's in every owner's (and I assume UHK developer's) interest to be able to definitively know whether it's a hardware or software issue.

mondalaci, 2018-08-07:
@linduxed Please do the same test as @kour1er with 50-50 settings. I'm interested about your findings. At this point, I don't have a better idea regarding testing UHKs.

kbranch, 2018-08-08:
So I'm a little late because I got sidetracked last night, but I did get to break out the oscilloscope and logic analyzer tonight.

I started by removing my 5 key so I could get a good look at the entire rising edge when pressing it.  I connected a 10k resistor between 3.3v and one side of the key and connected the other side of the key back to its regular spot on the board, where the controller can see it.  This made the controller think multiple keys were being pressed when I press the 5 key, but that's not a big deal for a test.

From there, I hooked the key up to an oscilloscope to see just how noisy key presses are.  I can take some screenshots if they're of any use, but they're not terribly interesting.  Most presses took around 0.3 ms to go from solidly low to solidly high.  The worst I ever saw was ~1 ms.  I'd guess that I looked closely at a few dozen presses on the scope.

Next, I hooked up a logic analyzer so I could line up Switch Hitter's detected chatter with what's physically going on.  I did my best to remember to set debounceTimePress and Release to 1 ms after power cycling.  This got kind of murky because detecting "chatter" in Switch Hitter does not necessarily mean anything physically went wrong at all, but I'll try to summarize what I learned on that front:
- Switch Hitter will detect "chatter" if you bump into the 6+4 simultaneous key presses limitation in the right way.  I was able to reproduce this pretty reliably by wildly mashing 6+ keys at once, at least on 8.4.0 and 8.4.2.  I'm not sure I tried on 8.3.3.  I don't know whether this is the firmware handling too many key presses in a less than ideal manner or just what happens when you roll your face on the keyboard.  Not a big deal in actual use, but it means I had to be a little careful when testing for chatter.
- Switch Hitter will detect "chatter" if you flutter around the actuation point, but I was able to verify with the logic analyzer that these were legitimate key presses that simply happened within the defined chatter threshold.  I never saw a case of this happening due to actual chatter.

And here are the conclusions that I'm reasonably confident of:
- My keyboard's green switches are physically well within spec - I never observed a transition that took longer than 1 ms, either with the logic analyzer or the scope.
- When carefully pressing <= 6 keys for longer than Switch Hitter's chatter threshold, I was unable to record any chatter at all on 8.4.0, either with Switch Hitter or the logic analyzer.  I can't say I'm 100% confident in this one, but I pressed a hell of a lot of keys like this.

I didn't test 8.4.2 for as long as 8.4.0 (I was on 8.4.0 when I started), but I thought it was also passing the "carefully press <= 6 keys" test while I had my key removed.  I think I'm noticing a little chatter as I type this with 8.4.2 and 5 ms debounce (press and release), though.  Before today, I had been using 8.4.0 for quite a while with the default 50 ms debounce without noticing any issues, but I'll play around with settings and versions as I use it regularly this week to try to narrow that down before I pop a key back off to try the logic analyzer again.

I also stopped testing 8.3.3 early because I thought I caught it red handed.  I swear I recorded it sending a key press that didn't register on the logic analyzer, but I must have misinterpreted the data - it's a legitimate key press.  I'll need to try again later, it definitely failed the "carefully press <= 6 keys" test, I just missed recording it.

Just let me know if you'd like me to try anything else next time, I don't mind fiddling with it to get to the bottom of this.

kbranch, 2018-08-08:
As I play with it today (without the logic analyzer), Switch Hitter has recorded extra key presses for keys that were definitely pressed for much longer than its threshold, using 8.4.0 and 8.4.2 with 5 ms debouncing.  It's rarer than I would have caught in my testing last night, I'll have to hook multiple keys up next time.

In the two examples I have so far, the extra keystroke actually happened on the release.  I have to admit that I was focusing on the key press last night - I took a few looks at the release without seeing anything nasty, but I'll have to look closer.

kbranch, 2018-08-09:
A little more testing with the logic analyzer, this time with some actual chatter captured.

I didn't feel like desoldering a switch again tonight, so I changed my tactics a little and set the logic analyzer to look for low pulses between 6 and 16 ms.  The firmware seems to poll every ~5 ms, so that means they key looks like it was released for only one or two polls before being pressed again.

I caught chatter on release twice - I've attached screenshots, Switch Hitter logs and the raw logic analyzer captures (use [this](https://www.saleae.com/downloads/) software to view the captures).

I think the next step is to desolder a key again and take a better look at those release edges.  A special firmware version for debugging that leaves the column lines high all the time may or may not help identify weird issues, though.
[ChatterOnRelease.zip](https://github.com/UltimateHackingKeyboard/firmware/files/2272718/ChatterOnRelease.zip)

Jopie01, 2018-08-09:
I digged a bit into the source and if @kbranch is right about the polling of ~5ms maybe there is some room for some changes. On Arduino I'm also using debouncers, but the polling speed is way faster (polling at 8 MHz, debouncing at 500 ms). To do so, I'm able to detect the rising or falling edge and start the debouncing right after. Also I get a more accurate result. So polling is running on 8MHz and when an edge is detected, the debounce start time is set. During the polling the debounce time is continuously checked. 
You have the debounce time now in the USB-reports, what if you put it in the keyscan function right?

Something completely different. I discovered that my switches seems to have two press states (maybe that's normal with mechanical switches). When I press the switch slowly around halfway through the character appears already on the screen. When pressing the key fully until it's stop, the character is repeated. Sometimes this happens, sometimes is doesn't.

kbranch, 2018-08-09:
The interesting part to me is that the release bounce is apparently lasting >= 5 ms in some cases (15 ms between visible pulses in one of the captures I posted).  Polling faster and using a smarter debounce algorithm might help, but this much bouncing seems unusual for Cherry switches.  I'm very curious to see the full release waveform.

mondalaci, 2018-08-12:
@kbranch Thanks for the detailed writeup! Very much appreciated.

The debouncer actually uses a millisecond timer, not a 5 ms timer, but I don't think it would make any difference.

You measured a release bounce that lasts for 15 ms which is larger than the 5 ms maximum specified bounce time of MX switches. This begs for some explanation.

I can imagine that the vast majority of the MX switches are within spec when they roll out from the factory, but a lot happens during their lifetime which can affect their bounce time. The switches get transported to a distributor, then to a manfufacturer where they're subjected to a lot of heat during wave soldering. Then they arrive to their final desination and get smashed by the user millions of times, experience temperature and humidity fluctuations, eat some dust, etc.

The bounce time of a switch is a result of its history. Based on a very small sample that @kbranch measured, it seems that the bounce time can be as high as 15 ms, but I'm sure it can be even higher.

As another example, I'm pretty sure something happened to @kour1er's UHK that resulted in extremely long bounce times. Being a pilot run UHK, my guess is that the operator of the selective wave soldering machine was tweaking process parameters, and the switches got too much heat and behaved funky as a result.

Given the above, I think the best strategy is to use the longest debounce time that doesn't cause usability issues. What usability issues? The UHK must be able to discern discrete keypresses, even when hit in rapid succession by humans. But how rapid is that? @kour1er send me a [crazy finger breaking game](https://www.sporcle.com/games/pr1nterbob/pressing-the-k-key) ealier that should give us a good idea. I can go slightly above 300 (that is, if 300 wouldn't be the upper limit in the game), maybe about 330, but I surely couldn't go above 400 and I doubt any human could go above 500 or 600.

300 hits per 30s translates to 100ms between hits, and 600 hits per 30s translates to 50ms which is the current default debounce interval, so I'm feeling quite strongly that the current value is close to optimal.

Also, based on your and my observations, the new debouncer is way better than the old one, and we didn't receive almost any complaints about the old one either, maybe only about 1-2.

When using the new debouncer with the default 50-50 debounce values, if bounces happen, I'd assume it's the switches, not the debouncer in which case a return is justified.

Feel free to let me know if you think I'm missing something, guys.

@Jopie01 @linduxed Do your UHKs still produce chatter?

[45 more replies]

PR #130 (2018-06-27): Fix key chattering
This resolves #128.
closed 2018-08-31 by mondalaci