UltimateHackingKeyboard/firmware #132: Wake up problem on Linux and Windows

mondalaci, 2018-06-30:
When I suspend Linux or Windows, unplug my UHK, replug it, and finally try to wake up the host by pressing a key nothing happens.

However, my UHK does wake up the host by pressing a key if I don't disconnect it from the host after suspend.

According to my testing, this issue persists since the inception of the wake up feature which was implemented by 3ab2ac18.

Lauszus, 2018-06-30:
This is most likely caused by this line never being called: https://github.com/UltimateHackingKeyboard/firmware/blob/e9cd3a96c2abbe92bdef6890c1264e3ba79c6520/right/src/usb_composite_device.c#L169, which it won't since the keyboard was not connected when the computer went to sleep. Thus this line is never called: https://github.com/UltimateHackingKeyboard/firmware/blob/d72ea4dde53af43b83d725dd4cc5f51404958837/right/src/usb_report_updater.c#L456.

We could simply call ```WakeUpHost(true)``` we could simply initialise ```IsHostSleeping``` to true.

Lauszus, 2018-06-30:
I just tested https://github.com/UltimateHackingKeyboard/firmware/issues/132#issuecomment-401548327 on Windows and that did not work.

However I tested another keyboard and that is also not able to wake up Windows if it is attached after Windows was put to sleep. I suspect it might not be possible and Windows do not init devices if they are plugged in when it is asleep.

Do you know of any other keyboard that is able to wake up Windows from sleep when plugged in afterwards?

Note that I have not tested this on Linux. My code is here: https://github.com/Lauszus/firmware/tree/sleep if you want to give it a try.

mondalaci, 2018-07-01:
I gave your code a try and it didn't make any difference. I could wake up the Mac but couldn't wake up the Windows or Linux machine. Same thing with a third-party USB keyboard. Given the above, this might not be possible at all.

What do you think? If you agree, I'll just close this issue as wontfix.

Lauszus, 2018-07-01:
I agree, seems like it is not possible on Windows and Linux.

Lauszus, 2018-07-01:
No, as it doesn't have any effect. It was just a quick test :)

mondalaci, 2018-07-01:
Thanks for confirming! Issue closed :)

closed 2018-07-01 by mondalaci