UltimateHackingKeyboard/firmware #136: The USB freeze bug is back

mondalaci, 2018-07-04:
@eltang @Lauszus I was shocked to notice that the freeze bug is still an issue in master. (Currently, master is at https://github.com/UltimateHackingKeyboard/firmware/tree/v8.3.3 )

According to my findings, when the USB test #ifdef clause is not used, the bug surfaces, but when it's used, the firmware is stable.

Now I'm testing the firmware with the test clause on, but with the `simulateKeypresses = !simulateKeypresses;` statement commented out and seems to be stable.

Any ideas?

mondalaci, 2018-07-05:
The freeze occurs during typing. I'm not sure how consistent the amount of time is because now that the test code is disabled it's significantly harder to reproduce, but I don't think it's very consistent. I hardly believe that sending reports during idle would solve this issue because the firmware seems to be stable when the test code is included, even if it's not triggered. I think it's some sort of low-level issue.

Lauszus, 2018-07-05:
Can you try to play around with the interrupt priority here: https://github.com/UltimateHackingKeyboard/firmware/blob/d72ea4dde53af43b83d725dd4cc5f51404958837/right/src/init_peripherals.c#L72

Lauszus, 2018-07-05:
@eltang the whole point of only sending the reports when the report changes is so it works properly with Windows. This is also how keyboard normally do it.

eltang, 2018-07-05:
Please refer to the documentation about HID idle [here](https://www.keil.com/pack/doc/mw/USB/html/_h_i_d.html).

mondalaci, 2018-07-05:
Thanks guys!

I was testing ef7589d which seems to be bug free.

In the meantime, I talked to @Lauszus and he said that this might be due to a GCC bug, and it's worth to play with the opimization flags. I switched to master, and flashed the debug build which shouldn't be optimized. So far so good. I'll be keeping you in the loop.

laxu, 2018-07-10:
Can confirm this is still happening on 8.3.3. Hopefully a fix is found soon as I can deal with having to reconnect it from time to time over not having working sleep and media keys.

mondalaci, 2018-07-28:
After testing the most recent firmware 8.4.0 for about 100 hours, I'm confident that this bug has gone.

Excellent work, team!

closed 2018-07-28 by mondalaci