UltimateHackingKeyboard/firmware #138: Make caps lock turn on by double typing shift

iprok, 2018-07-06:
Just as it's done now for Mouse,  Mod and Fn buttons it will be great if we can "lock" shift button. If we press shift twice it should make caps lock on.

mondalaci, 2018-07-06:
Thanks for the idea!

We only implement general features, and the generalization of this feature is to make modifiers lockable by double-tapping them.

However, this would introduce more problems than it solves. According to the feedback we receive, some users are outright frustrated by the double tap to lock layer feature because it locks layers unintentionally from time to time.

That's why we'll implement (actually largely implemented) per-key double tap to lock layer functionality, so that users will be able to enable/disable double tap to lock layer on a per key basis.

We could do the same for the double tap to lock modifiers feature, but based on the feedback we receive, I'm confident that the double tap to lock layer feature is much more often used than the double tap to lock modifier feature would be because nobody has ever requested the latter.

The main problem is that we'd have to add an additional user facing option for a feature which would be used by almost noone. This would make Agent harder to use for everyone whilst benefitting few.

I'm tagging this feature as an exotic feature and closing it. Such features will be reconsidered later. Feel free to follow up in the comments.

closed 2018-07-06 by mondalaci

kareltucek, 2018-07-06:
@mondalaci isn't the general modifier lock effectively solved by 617 in base layer?

Of course, shift-lock is apparently not same thing as caps lock... but I feel the nature of the problem is closely related...

I am also wondering whether requests like this could be solved by introducing some higher-level abstraction. (E.g., introducing double tap as a general trigger (e.g., like secondary-action-trigger and single-tap-trigger), and introducing layer toggle as an ordinary action (just like send-scancode-action))

I do not imply that this should be high priority issue. However, planning future implementation of exotic features via global generalizations seems much less problematic and more efficient in the long run than hot-fixing urgent-user-needs as special cases...

mondalaci, 2018-07-06:
https://github.com/UltimateHackingKeyboard/agent/issues/617 doesn't solve this one. It enables modifiers to have their own layers, but Caps Lock cannot be triggered this way.

As a developer, I appreciate the kind of abstractions that you suggest. It's nice to compose primitives to achieve anything. That's what https://github.com/qmk/qmk_firmware or https://github.com/keyboardio/Kaleidoscope do with hooks and plugins. However, I don't consider them to be a great solution usability-wise.

Agent is so easy to use because it offers practical features in an intuitive manner. By exposing every possible combination of every feature the UI would become vastly more complex, and most of the options offered woudn't make practical sense. I'd much rather never implement the least used 1% of features than overwhelm users with extra features they almost certainly won't ever use.

mondalaci, 2018-07-09:
@kareltucek Thank you for taking the time to create this very detailed writeup.

The locking of modifier layers will be possible, however the modifiers themselves won't be composed, only if the key action contains the modifier in question. This allows for greater configuration freedom.

Caps Lock by definition is not a modifier. It does not compose with other keys, so treating it like one doesn't feel right.

Regarding the UI, I can see your point, and I think you did a great job of laying out the single-tap, double-tap, and secondary role actions in the key action popover. I also understand the concepts of separating actions from triggers.

Yet, I think yours is a great example how introducing more features can make Agent less intuitive. The UI is more cluttered, even when double-tap action and secondary role action is collapsed, but the new content of the select2 boxes would be even more confusing as it'd overlap with the the features of other tabs of the key action popover. It'd also contain useless combinations like move mouse left on double tap.

I mentioned QMK and Kaleidoscope to exemplify composability via code. They don't have any reasonable UIs, indeed.

Regarding the rationale part your comment, I think yours is a critique of 60% layouts in general. This is hardware issue, not a software issue. We plan to design a 80% UHK in the future containing dedicated Esc, F{1-12} keys and a navigation cluster.

> I am just proposing a refactor decoupling actions from triggers.

You're talking about this like it's a simple thing. It is not. In practice, your proposal would blow up the whole firmware and Agent codebase. It'd affect everything very deeply and would take forever to implement. Then another 1% of users would eventually create further exotic feature requests, calling for another monumental refactor that would call for enormous work to create little value for a very small group while making things more complex for everyone else.

Please let's keep the conversation conscise going forward. It's extremely difficult to follow so many points.

kareltucek, 2018-07-09:
> Yet, I think yours is a great example how introducing more features can make Agent less intuitive.

I do not see that, sorry.

>the new content of the select2 boxes would be even more confusing as it'd overlap with the the features of other tabs of the key action popover. It'd also contain useless combinations like move mouse left on double tap.

Well, determining what exactly should be shown where is an implementation detail, which I think does not need detailed discussion at this time&place.

The example in question indeed has rather limited use. However, as long as the box is search driven, I generally do not see a problem in it. It already contains so many items that picking them by hand does not make sense and If the user explicitly asks software to do something, who are we to oppose him? 

> This is hardware issue, not a software issue.

Yes, definitely.

>We plan to design a 80% UHK

Great!

> You're talking about this like it's a simple thing. It is not. 

Well, I am simply discussing the points you keep putting forward as a problem, which is always the UI. 

> In practice, your proposal would blow up the whole firmware and Agent codebase.

Well, what should I say to this? I guess that claiming what you did right or wrong is not very constructive at this point. 

You are the master of your implementation and so you are the most likely one to come up with possible solutions to technical problems. I definitely see that this is a problem, and am neither claiming nor sure myself whether it is worthwhile. I can say only that I see the current state as conceptually wicked.

It is definitely up to you whether you decide to set this as a long term goal or not. I can only suggests that it usually helps to take some time to think of how to achieve things in the easiest possible fashion, rather than to take time to think up reasons why the things are impossible :-).

I definitely do not anticipate that you will agree to implementing things like this on a spot. It is rather a point for long term consideration in context of other problems and feature requests which pop up.

Honestly, I am also very curious about the macro engine which should be available soon, since seeing it will answer many question regarding actual strength and configurability of UHK.

mondalaci, 2018-07-09:
I'm not saying it's impossible. I'm saying that it's a huge amount of work that would result in a little gain for some and a net negative for everyone else.

I don't want to sound unconstrictive. Quite frankly, I'm rather overwhelmed nowadays, and can't sufficiently concentrate on this issue. Let's talk about it later.

kareltucek, 2018-07-09:
Sure

owlstead, 2018-10-22:
I'm all for making a less cluttered UI. However I side with kareltucek on thinking that double tapping shift is a great *and commonly used* way to switching on CAPS. All the keys on the base layer are basically the same as on *any* QWERTY keyboard *except the CAPS key* which has become the mouse key. So in this case UHK is the one introducing a one-off feature, not the users.

So personally I would seriously consider making double clicking shift a specific option for each key map. Leaving it on by default would probably break gaming though, so it should not be a default (!). Don't fuck with Strafe or you'll get whacked by a frustrated FPS freak :)

Otherwise yeah, enable double clicking to perform *any* action per key is the only configurable option that seems to make sense.

Just my 2cts. I cannot vote on features it seems, and I think the note about gaming helps talk about this "exotic" feature.

LazaroFilm, 2020-12-03:
Might I add that the iPhone uses double tapping the shift key for CapsLock. I know it's not a mechanical keyboard, but in term of number of user of the function you can't beat that. I would also like to add my voice to that feature request. Also, on iPhone it's double tap to activate CapsLock and single tap to deactivate it.