UltimateHackingKeyboard/firmware #140: Iinverted mouse button mode

Bysmyyr, 2018-07-14:
Mouse mode is awesome but  I miss one thing: possibility to use mouse button to disable mouse mode by pressing it when mouse mode is locked in that way that when I stop pressing it, the locked mouse mode goes back on. In other words inverted functionality when mouse mode is locked.

mondalaci, 2018-07-14:
Sorry, but it's unclear what you are after. What do you mean by "mouse button"?

luteijn, 2018-07-14:
I take that to be the key labeled mouse in agent. So, proposal is to make it possible to temporarily switch to the Base layer by holding the/a key that normally switches to that layer when another layer is locked in by double tapping. 

I feel it really comes down to generalizing the layers more, so there is nothing special about the 'base'-layer and one can freely switch directly to other without going back to Base layer first. Probably would tie into #617 , which basically also generalizes the idea of layers more.

Bysmyyr: a work around for you would perhaps be to just add a new keymap that basically has all the keys of the Mouse layer and Base layer reversed. Then just switch to that keymap when you would normally just lock the mouse layer on your current setting, and use the 'mouse' button to temporarily gain the ability to type a few characters.

mondalaci, 2018-07-25:
Thanks, now I see what this feature is about.

Quite frankly, I don't see this feature used by many, but it'd add to the complexity for everyone, so I'm tagging it as an exotic feature for further reconsideration and closing it in the meantime.

closed 2018-07-25 by mondalaci