UltimateHackingKeyboard/firmware #141: UHK wakes up computer from suspend to ram without key presses

mcscript, 2018-07-17:
Regularly (but not _always_, see observation below), when I suspend the computer to RAM using

```
systemctl suspend -i
```

(typed using the UHK), the computer will suspend, but immediately wakes up again. Here's the relevant output from `dmesg`:

```
[95028.217003] PM: Suspending system (mem)
[95028.217031] Suspending console(s) (use no_console_suspend to debug)
[95028.217338] sd 4:0:0:0: [sda] Synchronizing SCSI cache
[95028.217826] serial 00:01: disabled
[95028.217847] e1000e: EEE TX LPI TIMER: 00000011
[95028.231208] sd 4:0:0:0: [sda] Stopping disk
[95029.547266] PM: suspend of devices complete after 1330.086 msecs
[95029.565243] PM: late suspend of devices complete after 17.970 msecs
[95029.567165] xhci_hcd 0000:00:14.0: System wakeup enabled by ACPI
[95029.605198] PM: noirq suspend of devices complete after 39.950 msecs
[95029.606090] ACPI: Preparing to enter system sleep state S3
[...]
[95030.433434] ACPI: Low-level resume complete
[95030.433555] PM: Restoring platform NVS memory
[95030.434304] Suspended for 2.912 seconds
[95030.434351] Enabling non-boot CPUs ...
```
This only happens if the UHK is attached. It does not happen with another keyboard (Cherry MX 3850 USB).

Additional information:

OS: Linux 4.9.0-6-amd64
UHK firmware: 8.3.2

**(Possibly) important observation:** I have observed that, _if_ the computer properly suspends, the LED display of the UHK is **not** lit. In case I experience the above-mentioned issue, the display always seems to stay lit.

This might be related to #85, although this is marked as fixed and I am using firmware 8.3.2.

Please let me know if I can assist in debugging this issue.

mondalaci, 2018-07-17:
I'd also think that it's related to #85. Firmware 8.3.2 does not contain the relevant fix due to "Revert the Windows related commits of firmware 8.3.1 because they introduced a nondeterministic bug that made USB hang."

Please give a try to the (potentially unstable) firmware 8.3.3 which contains the relevant fix, then downgrade back to 8.3.2.

mondalaci, 2018-07-25:
Just released firmware 8.4.0 which seems stable and should fix this issue. I consider this bug to be resolved unless otherwise noted.

mcscript, 2018-07-26:
Please re-open this issue (I do not have a button to do so).

In terms of suspend to RAM issues, I got a similar behavior as with 8.3.3: Out of 2 suspends so far, 1 immediately woke up the computer. LEDs on the UHK were still lit.

mcscript, 2018-07-26:
[Edited my last comment to make clear that these new observations have been made with firmware 8.4.0]

mondalaci, 2018-07-26:
When you suspend you computer, I assume that the longer you press the Enter key of your UHK at the end of the referenced command, the more likely it is that your UHK wakes up your computer. If you press it momentarily, it will likely not wake up your computer. Can you check if this is the case?

luteijn, 2018-07-26:
if releasing in time is an issue, try 
sleep 2; suspend-command <enter>
to give you time to release the key..

linduxed, 2018-07-31:
While I've never experienced the issue before, as of `8.4.0`, this issue has started to be a problem for me as well.

mondalaci, 2018-07-31:
@mcscript @linduxed Please try the newly pre-released firmware 8.4.1 and let me whether it fixes the issue.

linduxed, 2018-07-31:
Updated the firmware. The first suspend resulted in an immediate resume. All the following ones worked as intended. I'll report in a day or two if things are working as expected.

Maybe interesting detail: prior to `8.4.0`, pressing keys on the keyboard did not wake the computer. With `8.4.1` pressing a key on the keyboard results in a wake-up.

mondalaci, 2018-07-31:
@linduxed Pressing any key on your UHK keyboard should wake up your computer. This is the intended behavior.

How do you exactly put your computer to sleep? Via pressing a key(s) on your UHK. Please describe in detail.

My guess is that you use a sleep key on your UHK, and your computer goes to sleep very quickly and by the time it sleeps, the key on your UHK is still pressed, and your UHK sends a wake up event to your computer, so that it wakes up.

luteijn, 2018-08-01:
If this is the case, the workaround is probably to just write a macro that 'taps' the sleep key, and assign that to the sleep key. Since macro's don't autorepeat, they can release keys pretty quickly and you could even add a small delay before sending the command to make sure...

mondalaci, 2018-08-01:
@linduxed Does sudo ask for a password? If so, the sleep doesn't have any practical effect, as you may be  holding the last key of the password while your computer goes to sleep.

linduxed, 2018-08-01:
This particular command is allowed in my `sudoers` file.

mondalaci, 2018-08-01:
@linduxed How often does your UHK wake up your computer? According to the above, it seems nondeterministic.

mondalaci, 2018-08-10:
@mcscript When saying that suspend attempts failed, do you mean that your computer momentarily suspended but immediately resumed? Maybe you weren't able to notice it in which case the kernel log may give you a hint.

Please report the lockups in the relevant issue and provide the output of the suggested script. That way, we can actually make progress.

mondalaci, 2018-08-12:
@eltang I think we should simply ignore keypresses for a sec right after suspend. That way users wouldn't resume accidentally.

mcscript, 2018-08-13:
@mondalaci Yes, If I say it "fails", I mean it according to the original report, i.e. the computer suspends, but immediately wakes up again.

closed 2018-08-13 by mondalaci

mondalaci, 2018-08-13:
Fixed in firmware 8.4.4

PR #153 (2018-08-13): Don't wake the host if a key is held down through the beginning of sleep
This closes #141.