UltimateHackingKeyboard/firmware #145: keymap switching activation from non-default layer via secondary action

kareltucek, 2018-07-25:
Mapping:
- mouse layer as a secondary action of a key (let's say the 'mouse' key)
- keymap switch on another key (let's say 'A' key) in the mouse layer

Expected behaviour:
- <mouse + tap A> switches to another keymap

Actual behaviour:
- <mouse + tap A> lights up the mouse indicator, but does not switch the layer
- <mouse + double tap A> switches the keymap

kareltucek, 2018-07-25:
I consider this a low-priority issue (as well as #146), however, the behaviour is confusing, especially when combined with #146, since then the keymap seems to have hanged. (One use case where the two issues may collide is cycling through  keymaps via a single shortcut.)

mondalaci, 2018-07-25:
This is a duplicate of #69, but instead of removing this, I rather remove #69 because this one is better written.

PR #149 (2018-08-02): Fix various secondary role bugs
This closes #145, closes #125, closes #119, closes #117, and closes #69.
closed 2018-08-02 by mondalaci