UltimateHackingKeyboard/firmware #146: Secondary key role not triggered after switching keymap

kareltucek, 2018-07-25:
Mapping of keymap A:
- mouse layer mapped as primary action of the mouse key
- switch to layer B mapped to '+' in mouse layer

Mapping of keymap B:
- mouse layer mapped as secondary action of the mouse key
- some key with actions in base layer and mouse layer

Expected behaviour (starting in keymap A):
- press and hold 'mouse'
- tap '+' switches to keymap B
- tap <some key> produces its mouse-layer action

Actual behaviour (starting in keymap B):
- press and hold 'mouse'
- tap '+' switches to keymap B
- tap <some key> produces its base-layer action

kareltucek, 2018-07-25:
[UserConfiguration146_2.txt](https://github.com/UltimateHackingKeyboard/firmware/files/2228305/UserConfiguration146_2.txt)

- Use mouse+'+' to cycle into BGA layer. 
- Press and hold 'mouse'
- tap A (switches into BGB layer)
- tap A again (should produce 'b', but produces 'a' instead)

Layers BGA and BGB contain a simplified test case. Also, you will notice the bug during the first step, because you will have to depress 'mouse' several times.

I am using the 8.4.0 pre-release firmware.

kareltucek, 2018-07-26:
@mondalaci I'm pretty sure it finally *does* switch to BGA via two or three other layers. The catch is that you have to depress the mouse button several times during the process (i.e., cycling mouse + doubletap '+' + depress mouse). This is caused exactly by combination of the two bugs (#146 + #146/#69).

The keyboard abbrevs in order you should see are MRR, ARR, TES, TS2, BGA, BGB. First hang comes on TES, where you have to depress mouse, press it again and double tap '+'.

mondalaci, 2018-08-19:
I'm sorry, but we can't make this happen, because #157 called for a behavior that is in conflict with this one. The behavior is to suppress keys upon layer changes. And given that keymap changing changes the function of keys just like layer changing does, the behavior should be consistent.

closed 2018-08-19 by mondalaci

kareltucek, 2018-08-19:
Ok, now I realize that the specific nature of secondary roles would lead to triggering action on key depress in case of trivial restoration of the state. Still, functionally and conceptually I see no conflict. So I guess a solution is extending the state enum so that it contains a nontriggering state?

eltang, 2018-08-19:
@kareltucek The solution for #157 was to suppress pressed keys as part of the layer or keymap switching process, which means disabling their action until their next press. A future change could make it so that they still produce an action after a layer or keymap switch, but the action is taken from the layer which was active when they were first pressed. The exception for modifiers is a stopgap measure to emulate this behavior before the change happens. But the main idea is that keys are to be prevented from taking on any new action after a layer or keymap switch and before their next press. In addition, a future Agent update may make it so that different keymaps don't even have the same number of layers. I hope you can see why it would be difficult to accommodate your case.

kareltucek, 2018-08-19:
Thank you for trying to explain, but I am still a bit lost in your train of thought.

> The exception for modifiers is a stopgap measure to emulate this behavior before the change happens.

I am very sorry, but, I do not understand this sentence.

> The solution for #157 was to suppress pressed keys as part of the layer or keymap switching process, which means disabling their action until their next press.

I totally agree with that as long as 'actions' means sending scancodes (or mouse clicks and movements).

> But the main idea is that keys are to be prevented from taking on any new action after a layer or keymap switch and before their next press. 

When you are talking about actions, do you actually mean 'roles' of modifiers and layer switchers too? 

I believe that correct behaviour is to suppress sending scancodes upon layer change, but to activate modifiers according to the newly switched layer (or keymap). Primary layer switchers currently behave according to this logic. Modifiers (shift, ctrl, etc) also behave according to this logic. Secondary layer switchers do not. 

> I hope you can see why it would be difficult to accommodate your case.

Honestly I do not. I have a strong feeling that there is a misunderstanding between us regarding what the case is.

> In addition, a future Agent update may make it so that different keymaps don't even have the same number of layers.

There's no problem with that. I could see a problem in resolution if the firmware allowed nested layer switching, but it does not.

eltang, 2018-08-19:
> 

@kareltucek 

>When you are talking about actions, do you actually mean 'roles' of modifiers and layer switchers too?

An action is simply a key doing something. This includes everything.

>I believe that correct behaviour is to suppress sending scancodes upon layer change, but to activate modifiers according to the newly switched layer (or keymap). Primary layer switchers currently behave according to this logic. Modifiers (shift, ctrl, etc) also behave according to this logic. Secondary layer switchers do not. (And I do not see reason why.)

It is precisely this kind of behavior which #157 aims to prevent. Changing a layer or a keymap should not result in an already pressed key to suddenly do something different.

The current implementation is not perfect yet. Basically, the target behavior will be something like this. When the layer is switched, any keys which are pressed at that time will behave as if the layer switch did not happen until their next pressed (this includes the modifier case). By contrast, when the keymap is switched, any keys which are pressed at that time are suppressed, or in other words do nothing, until they are pressed again. In both cases, the firmware would not discriminate between different types of actions.

kareltucek, 2018-08-19:
I've just noticed one thing in the provided keymap, which might have led to some confusion: the 'a' letter mapped on the 'mouse' button is there only to be there (because agent does not allow empty scancode). It is the secondary role which matters for it. The 'a' in the expected output should be produced by the physical 'a' key (i.e., the key which has 'a' printed on keycap), not by the mouse button.

kareltucek, 2018-08-19:
> The current implementation is not perfect yet. 

Ok, I see what you mean now. (Although I am not sure how much I agree/disagree.)

The fun fact is that then the secondary role is the only case which works according to specification.

kareltucek, 2018-08-23:
@eltang I assume you are aware of this, but in case... On 8.4.5., layers activated by secondary roles are *not* deactivated on change of keymap (which I believe is contrary to the described target behavior).

eltang, 2018-08-25:
@kareltucek Thanks for the report, I didn't get a chance to test this before 8.4.5 was released. The next release will fully implement the target behavior.