UltimateHackingKeyboard/firmware #150: Modifier keys doesn't stick on double press

iprok, 2018-08-05:
I have upgraded to firmware 8.4.2 and I've just notified that not mouse nor Mod key doesn't stick anymore when double pressing it.

mondalaci, 2018-08-06:
Chances are you have to double tap more quickly. Please give it a try.

Which was your previous firmware version?

mondalaci, 2018-08-06:
Ultimately, we'll make the double tap to lock timeout configurable via Agent but it'll take some time because it'll involve changing the configuration format, and increasing the hardcoded timeout would make people accidentally lock Mod, which would mostly do a disservice usability-wise.

As the first step, Agent will make it possible to configure the the double tap to lock feature on a per key basis. This is however dependent on #148, so we have to fix it first. Afterwards, we can increase the hardcoded timeout, and eventually, it'll be configurable via Agent.

Closing this, as it's a redundant issue.

closed 2018-08-06 by mondalaci

mhantsch, 2018-08-13:
I guess this is a duplicate of #749. Putting this comment in here for easier reference to that issue.