UltimateHackingKeyboard/firmware #151: Unable to make a text selection with the mouse feature with shift and two clicks

richrd, 2018-08-08:
Text selection works fine when clicking and dragging, but not without dragging.
With a regular mouse I can click where I want to start a selection, then hold shift and click somewhere else. 
That results in the text between the clicks getting selected. I can't seem to do that with the UHK mouse feature. I'm not sure if it's a firmware bug but I thought I'd report it here just in case.

kareltucek, 2018-08-08:
Shift is not bound to shift key in the mouse layer by default. Bind it via agent and it will work as expected.

kareltucek, 2018-08-08:
Definitely.

mondalaci, 2018-08-08:
What a fabulous oversight on our part! Thanks so much for reporting!

Just fixed this in https://github.com/UltimateHackingKeyboard/agent/commit/80e8c014ec493378bde851c2ebfe4f906fbe2487

(The firmware builder uses Agent to generate the binary configuration that gets transferred to the EEPROM, so the above commit fixes this issue.)

closed 2018-08-08 by mondalaci