UltimateHackingKeyboard/firmware #152: secondary layer switch not working if next key also has secondary layer switching on the base layer

mhantsch, 2018-08-10:
Firmware 8.2.5, Agent 1.2.5.

Configuration - Base Layer
CapsLock: primary = keypress "Backspace"; secondary = layer switch to Mod layer
Left Space: primary = keypress "Space"; secondary = layer switch to Mouse layer

Configuration - Mod layer
Left Space: keypress "Enter" (no secondary function)

Configuration - Mouse layer
CapsLock: keypress "Enter" (no secondary function)

When I hold down CapsLock and press Left Space, I would expect an Enter keypress. Instead, nothing happens, and when I release CapsLock I get Backspace.

When I hold down Left Space and press CapsLock, I would expect an Enter keypress. Instead, nothing happens, and when I release Left Space I get Space.

I am not sure if this has already been fixed as part of #149 or #125, but want to record it for now so it can be retested.

mondalaci, 2018-08-10:
Please always test the most recent firmware version which is currently 8.4.2. I'm closing this assuming that this has been fixed, and will reopen if needed.

mondalaci, 2018-08-10:
As a rule of thumb, please always report issues against the latest firmware version. As you can see, the referenced issues are already closed.

mhantsch, 2018-08-10:
Well, you now got me to switch to the prelease / alpha versions, download and update to firmware 8.4.2, and the issue still occurs.

closed 2018-08-13 by mondalaci

mondalaci, 2018-08-13:
Fixed in firmware 8.4.4

mhantsch, 2018-08-14:
Yes, I can confirm this is fixed in 8.4.4.

PR #154 (2018-08-13): Ensure that secondary roles are triggered consistently
This closes #152.