UltimateHackingKeyboard/firmware #155: Agent hangs/becomes slow when Visual Studio debugger is running

roel-v, 2018-08-17:
When the Visual Studio debugger is running (so when I have a program running under the debugger), Agent becomes very slow to the point where the UI stops responding. Also I get occasional 'Reading your configuration' dialogs which then never finish. Only option it seems is that I have to stop all debugging instances, close or force kill all agent instances, then restart.

mondalaci, 2018-08-17:
Please update to firmware 8.4.4 and let me know if you still experience the stuck key issues.

We've never heard about the Visual Studio debugger issue. Is it Visual Studio Code? I'm asking because both Agent and Visual Studio Code are electron applications, so this may have something to do with it. If not, which Visual Studio Code version and which OS version?

roel-v, 2018-08-17:
Well, this has not been fun...

So I tried to upgrade the firmware. My computer is in the basement/server room and everything is hooked up on my desk via a USB3 hub (with a 20m USB extension cord). When I tried to do the update, I got a bunch of error messages, Agent became unresponsive, the keyboard became unresponsive, and every time I moved my mouse (!) the computer beeped. Unplugged the keyboard, mouse, other USB peripherals - then nothing worked at all any more (mouse nor keyboard), except for a beep every 2 or 3 seconds. (so I couldn't save the log of the flash diagnostics because the keyboard and mouse didn't work any more.)

I can see why devices connected via multiple hubs could be more difficult from a software point of view, but it all worked fine with my Ergodox, including flashing new firmware.

Hard reset computer, won't boot. Windows (10) gets to 'loading' screen (with the Windows logo, just before the spinning circle underneath the logo appears), then does nothing. Windows repair starts after a few reboots. I unplug the UHK, plug in my Ergodox, everything works. Reboot with UHK plugged in, same problem. But when I unplug the UHK at the Windows boot screen, Windows reboots itself and gets to the login screen.

So, I dig out an old laptop, put the UHK + all software on it (suggestion: put a big bold 'download' section at the top of the website, which links to (for the detected OS) Agent + firmware; it's a pain digging it up now, I have to resort to Google every time). I flash the firmware, fails after the right side is done (I do have a log of this one if you want). The leds on the UHK just sit there flashing (the letters), I'm wondering when it will finish, let it sit for 15 mins, unplug, retry, same result, hard reset UHK, get scared that I'll lose all my customizations (where is it documented what happens when you hard reset the keyboard? Somehow it got my settings back after doing the reset, but I'm not sure how), finally use another USB port on the laptop to plug the UHK into, then upgrading both halves works. The first USB port works fine when transferring files off a USB stick etc., so I'm not sure what the problem was.

Anyway, reconnected UHK to my main machine (when it was already running and booted into Windows), rebooted, this time at least it booted Windows correctly. So I hope that my boot issues are gone with this firmware.

I have yet to check on the issues that started this whole ordeal. It's regular Visual Studio, not VS Code - VS 2015 on Windows 10, I can't easily check if it's also a problem with VS2017.

Furthermore, may I suggest adding an 'auto backup' to Agent (which stores the current configuration, in a file with a timestamp in its name, so that you automatically have a history of settings), and renaming 'Export device configuration' to 'Back up UHK configuration' and 'Import device configuration' to 'Restore UHK configuration backup'. Also everywhere where Agent says 'Device' in the UI, it should just say 'UHK' or so. Oh and hide the 'LED brightness' section if it can't do anything anyway, but I already complained about that somewhere else earlier this week.

Now that I'm piling disjoint issues into this one ticket anyway, the 'Test' keymap shouldn't be showing to regular users, either, and the firmware upgrade should show some sort of progress bar and/or signal when it's done or in a stuck/failed state.

Lotharyx, 2018-08-17:
On 08/17/2018 09:05 AM, roel-v wrote:
>
> Well, this has not been fun...
>
> So I tried to upgrade the firmware. My computer is in the 
> basement/server room and everything is hooked up on my desk via a USB3 
> hub (with a 20m USB extension cord).
>
A comment from the peanut gallery: 20 meters is way longer than the USB 
3.0 spec maximum cable length of 3 meters.  Are you using an active 
extension cable?  If not, I am surprised it works at all. Extending USB 
over that distance is probably not the best solution to your need to 
have a console that far away from the computer.  You might want to 
consider using some network KVM option such as remote desktop, or at the 
very least, an Ethernet-based USB extender.

At any rate, I'd suggest your first action is to try everything with a 
much shorter cable and see if it works.

roel-v, 2018-08-17:
Yes it's an active cable - but upon checking I think it's only 15m. Either way, it has worked great in many difficult circumstances for the last almost 3 years, including high speed data transfers, access USB devices from emulators and networked-debugger-mobile devices, having many devices being plugged in/out, boot from usb etc. I understand that complex USB topologies are more difficult to implement correctly, but I've seen it all work before.

mondalaci, 2018-08-17:
@roel-v I'm sorry about your experience. Things are unusually problematic on your end, and I'm not sure what's the cause of all these issues. These haven't been reported by anybody else.

1. Did you manage to update both halves to firmware 8.4.4?
2. If so, did this resolve the stuck keys issue?
3. Does Visual Studio still cause problems after the firmware update?

roel-v, 2018-08-17:
Yes both halves are now at 8.4.4. I'm not sure yet if I'm still seeing the original issues, I'll update when I've tried more (although it seems to be fine at first sight).

roel-v, 2018-08-17:
Well I've found an issue, but I'm not sure if it's the same as before (some symptoms are the same). When I unplug the UHK (which I had to do because I accidentally activated the 'Test' keymap, and it seems there is no way to get out of it?), the whole machine became sluggish, mouse jumped, speaker beeps like described before, then a few seconds later all usb devices stopped responding all together. Hard reboot computer, all works fine. I'll see if I have some time this weekend to reproduce. Is there a way to capture the state of the USB subsystem? It seems like the USB stack gets confused at some point, like stuck in a loop or deadlocked while retrying getting a handle or something.

mondalaci, 2018-08-21:
@roel-v Given your recent comments, I think chances are good that it's resolved, so I'm closing this issue. Feel free to comment and I'll reopen it if necessary.

closed 2018-08-21 by mondalaci

roel-v, 2018-08-22:
I'm not sure if this should still be under the same issue, but it feels like it's all connected somehow, although that's only a hunch.

I haven't tried the unplugging-freezes-machine yet, but with the new firmware, every say 10 minutes, my headphones disconnect and reconnect (the other issues seems to be gone though, so that's the good news). These are Logitech USB wireless headphones, so it seems like every couple of minutes, the whole USB stack is restarted or reset or something similar (I've never programmed low level USB things, I'm not very familiar with how it all works). So just like the previous issues, intuitively it feels like the keyboard is sending either too many or unexpected values, and that something on the OS end gets jumbled up over it. What do you use for USB sniffing? Maybe if I could monitor USB traffic, I can find a relation between events on the keyboard and the host OS doing weird things.

mondalaci, 2018-08-22:
I don't believe that the UHK is the culprit. We've sent out more than a thousand UHKs, and nobody has reported such symptoms. Please connect your UHK directly to your computer, not via the 15m long USB extension cable and let me know what happens.