UltimateHackingKeyboard/firmware #156: Dev docs refinements

kareltucek, 2018-08-17:
I've read through the docs of the three main repositories (agent, firmware, bootloader). Yet, I find some points unclear (with respect to minimum configuration needed for firmware tweaking of a factory set-up UHK):

- I assume that flashing (locally built) bootloader onto UHK with factory bootloader is not necessary, right? 

- Furthermore, I assume that the J-link probe is not necessary for higher-level firmware hacking?

- When I open uhk-right project, MCUxpressoIDE shows error `Unable to find part support for NXP MK22FN512xxx12 in project uhk-right Resetting to generic part for core null	uhk-right		Unknown	Problem`. However, it can build the project fine.

- If I ignore the bootloader and j-link probe, the dialog in the step `Then in KDS, click on Run -> Run Configurations, select C/C++ Application -> uhk60-right_release_kboot, and click on the Run button.` shows error `Program not specified`, apparently requiring specification of the 'C/C++ Application' field (which is empty).

mondalaci, 2018-08-21:
I think now that I updated the instructions this should be resolved. Feel free to comment nevertheless.

closed 2018-08-21 by mondalaci

kareltucek, 2018-08-24:
> I've just updated them. 

Thanks very much! Instructions now see clear to me and I can flash from IDE.

> Not sure about the "Unable to find part support..."

Since everything seems to work, I would probably leave it be (unless you are interested in the output?). 

> What do you want to tweak in the firmware exactly?

No exact plans yet. So far, I've been doodling with layer switching code and macro player - just to find my way around the code of firmware and agent. 
The following points are currently of interest for me:

- Generalization of layer switching I.e., 
  - Refactoring switching so that the base layer is no longer special. This means allowing toggling layers from non-basic layers and activating one "hold" layer from any toggled layer.
  - Maybe allowing nested activation of "hold" layers.
  - (The idea is that agent's side would allow advanced mappings only with a special flag (like "advanced features" checkbox somewhere), altering firmware's behaviour only on those serialization cases which are currently not used at all by the agent.  )

-  If #157 turns up to be a problem, then I might need to deal with it somehow. (So far, I don't mind the behaviour (as long as it is consistent) with respect to use-cases which I can foresee. However, I still fear that the logic might become problem under some circumstances.)

- Adding some functionality to macro engine - I am missing layer/keymap switching. 

- Also, adding some generic support for adding arbitrary firmware-sided functionality via macro engine would be nice (i.e., coding agent so that adding new firmware-sided features is not so painful).

Should I fire these as separate feature requests? 

My current idea is to allow bindings that smooth out the workflow - i.e., to switch layers implicitly on actions. (A bit like vim's modes - the idea is to use only commands and let the editor (keyboard) switch layers according to their context.) 

Btw is this style of doc/dev questions/requests preferred or should I get in touch directly?