UltimateHackingKeyboard/firmware #158: Mod, Mouse, Fn and LED display turn on while I'm asleep

Hurricaaane, 2018-08-17:
I don't know how to formulate this issue report, if it is even one. I'm using firmware 8.4.2.

When I shut down my computer but keep it connected to mains power, the LED display and other indicator lights are all turned off.

When I wake up in the morning, I find that the Mod, Mouse, Fn and LED display are lit, the lock/agent/adaptive icons are not.

It has happened several times already. Note that I don't really care about this issue myself, but I wanted to point it out anyways in case it's relevant, I'm surprised this can happen at all without me touching the keyboard for more than 7 hours.

Hurricaaane, 2018-08-18:
Windows 10 Professional, it is a desktop machine.

Again I don't really mind  about this issue, but I'm reporting it anyways in the slightest chance that this bug report might be related to the freezing bug -which by the way I have not experienced so far with this firmware- or something else.

mondalaci, 2018-08-18:
I realized that there's a far easier way to reproduce this issue. I just put my Windows 10 laptop to sleep, and plug in my UHK. Then the layer LEDs immediately lit up.

This bug was introduced in 8.4.1, and more specifically, according to git bisect 2d7cd6845937c6dc0ce5bb61d07b394868227549 is the culprit.

I think this is hardly related to the freeze bug, but we should fix it regardless.

Jopie01, 2018-08-20:
I'm on Fedora 28, and have the same behaviour. I connect my PC to the mains and press the start button to start the PC. All the LEDs on the UHK lit up. After the BIOS has initialized the USB controllers, the LEDs are switched off.

owlstead, 2018-11-10:
This _seems_ a relatively easy to fix bug. Why is it still open?

mondalaci, 2018-11-10:
@owlstead Because our firmware developer has not been available lately. Please don't bump issues, because it only creates noise, and doesn't get them fixed sooner. You're welcome to send a PR.

owlstead, 2018-11-11:
@mondalaci apologies, removed comment. This comment will self destruct.

mondalaci, 2020-05-03:
@Hurricaaane @Jopie01 Can you reproduce this with the latest 8.9.3 firmware?

kareltucek, 2020-05-03:
(@mondalaci does this mean that you can no longer reproduce it according to your post above?)

mondalaci, 2020-05-03:
@kareltucek Correct.

owlstead, 2020-05-04:
I think I bumped this issue once. I vaguely remember seeing this, but for me the issue has disappeared. I'm using both Windows and Linux, but mainly Linux currently - and on a different machine. So possibly that's not saying _that_ much.

mondalaci, 2020-05-04:
Thanks for letting us know! Assuming has became a non-issue, I'm closing it.

closed 2020-05-04 by mondalaci