UltimateHackingKeyboard/firmware #161: [mac] caps lock light not turn on

kuil09, 2018-08-24:
hi, my keyboard's caps lock led not turn on when I using on MAC.
But on windows, it works perfectly.

my firmware version is 8.4.5

mondalaci, 2018-08-24:
It's probably not your UHK, but your Mac is misconfigured or running some application that affects caps lock. You can check caps lock state with another USB keyboard. Please confirm that it's not a keyboard issue, and once you got the solution, please let me know.

mondalaci, 2018-08-26:
@kuil09 I think you have Karabiner Elements running and the "Manipulate LEDs" option is configured as such that caps lock is not shown on USB keyboards. Please confirm.

kuil09, 2018-08-26:
I'm not using Karabiner but I refigure out whats going on here.

mondalaci, 2018-08-26:
@kuil09 This must be a Mac specific issue.

Please let me know what causes it when you figured it out.

kuil09, 2018-08-26:
Thansk to your fast response. Have a good day.

kuil09, 2018-08-26:
@mondalaci 
Oh, It was just my mistake. 
It effected from layout for "Windows" and layout for "Mac".
Interest thing is capslock work perfectly just LED not working on different layout.

mondalaci, 2018-08-26:
@kuil09 Can you please explain in detail which OS setting was wrong?