UltimateHackingKeyboard/firmware #165: Issue with key combos as primary with a secondary roles assigned

kour1er, 2018-08-28:
Firmware after 8.4.1 (i.e. 8.4.1 doesn't exhibit this behaviour) seems to be having an issue if the primary role of a key is a combination and the secondary role a modifier.

For example: if I have the right alt key (that is to say the 3 key in on the bottom row on the right side) set to send super + space (that is to trigger Spotlight search on a Mac) and the secondary role to send right alt --- the super + space (the primary role) doesn't seem to be sent consistently. 

Occasionally it sends the combination but other times only seems to be sending space. Almost as if the combination is being sent too fast to register to the operation system?

If I revert to 8.4.1, I don't have the issue anymore.

mondalaci, 2018-08-28:
I can clearly reproduce this issue. I'm on 8.4.5 and only the scancode gets sent and no modifier when tapping. Assigning this to @eltang.

closed 2018-09-03 by mondalaci

kour1er, 2018-09-14:
Just to provide another data point, I've just updated to the latest 8.4.5 and am still seeing this issue. Rolling back to 8.4.1 fixes it.

PR #168 (2018-09-03): Send primary role modifiers consistently
This closes #165.