UltimateHackingKeyboard/firmware #167: Don't suppress keys upon keymap change

mondalaci, 2018-09-01:
Let's say I map fn+\ to toggle between several keymaps. I used to be able to hold fn and tap \ several times and the display updated, but starting from 8.4.5 i have to release fn every time.

Please restore the old behavior regarding keymap switching but keep the new key suppression behavior regarding layer switching.

mhantsch, 2018-09-06:
Yes, I was just about to create an issue for this. I use FN+ISO key to rotate (cycle) between three keymaps. Each keymap maps that key to switch to the next keymap in the cycle.

Since firmware 8.4.5 this is no longer possible to cycle the keymaps by simply holding down FN and tapping ISO several times until the wanted keymap shows up on the display. It used to work before 8.4.5.

mhantsch, 2018-09-18:
I switched back to firmware 8.4.4 because of this issue. This allows me to cycle through my keymaps with Fn+ISO while keeping Fn held down. 

I have, however, noticed another behaviour in 8.4.4 which is related to this. Let me describe the relevant parts of setup first:

Keymap "CMX": Left Fn is standard Fn modifier, Right Fn is standard Fn modifier. On Fn layer, ISO key switches to "NMX" keymap.
Keymap "NMX": Left Fn is standard Fn modifier, Right Fn is standard Fn modifier. On Fn layer, ISO key switches to "NM2" keymap.
Keymap "NM2": Left Fn is standard Fn modifier, Right Fn is "," and its secondary action is Fn modifier. On Fn layer, ISO key switches to "CMX" keymap.

When using Left Fn + ISO, I can cycle through all keymaps by keeping Left Fn held down and repeatedly tapping ISO: CMX->NMX->NM2->CMX->...

When using Right Fn + ISO, this only works until I reach the NM2 keymap. The next tap on ISO will not cycle anymore. I think this is because the secondary action on Right Fn does not trigger if the key is already held down when the keymap becomes active. Interestingly enough, when I push Right Fn down and _start_ the cycle on the NM2 keymap, then I can cycle as much as I want (as long as I keep Right Fn depressed and just keep tapping ISO). But if I start that same cycle on any other keymap, then the cycle stops when I reach the NM2 keymap.

I guess I am saying that when you work on this issue, please also make it work in the case where the modifier is on a secondary action.

kareltucek, 2018-09-18:
Yeah, this was already reported on 8.4.4 in #146. (the current behaviour was actually implemented as a fix of that and one other related issue).

mhantsch, 2018-09-18:
Thanks @kareltucek -- I must have overlooked that one (maybe because it was already closed).

kareltucek, 2018-10-06:
Nope, 8.5.2 still does not work properly with respect to secondary roles. 

Consider the following UserConfiguration
[UserConfiguration_km_cycling.json.txt](https://github.com/UltimateHackingKeyboard/firmware/files/2453357/UserConfiguration_km_cycling.json.txt)

"Mouse" + "+" (hardware keys) should cycle through "MA", "TST", "TS2", "TS3".

kareltucek, 2018-10-06:
Oh, sorry, forgotten I have switched layers. Disregard the previous comment please.

kareltucek, 2018-10-06:
Still, Using the shortcut should bring me to TST. Then, pressing and holding "Mouse" again and tapping "+" should bring me to the "MA" again before breaking. However it breaks on "TS3" so it indeed does not work consistently.

So what behaviour is currently the one considered correct? Suppressing upon keymap change, not suppressing upon keymap change or suppressing upon keymap change and applying the new keymap actions?

eltang, 2018-10-06:
Use case 3 will conflict with #157, so I think this needs to be given a little more thought. As for use case 2, I don't see any risk of accidental activation of the primary role, as it is the secondary role which will be activated after the timeout.

mondalaci, 2018-10-07:
You're right regarding case 2. I didn't think it through. Let's make it happen then.

Regarding case 3, I can see how #157 conflicts with this but I think it'd be worth to make an exception in this case. No scancodes will be accidentally emitted in this case, after all.

kareltucek, 2018-10-07:
Personally, I think that most importantnly the target behaviour should behave according to a simple & generic ruleset. Currently I have no idea which ruleset is being applied. I am just quite confident that in the end, cases 1 and 2 should work consistently because their rulesets should be consistent due to their abstraction level relatedness. The case 3 should be considered in the bigger picture but may or may not be made work in the end.

I see three options what to do on keymap change:
1. Suppress everything. This works consistently in all three cases.
2. Supress nothing - i.e., keep the old action until the key is released. This works in cases 1 and 2 and is consistent with layer switching rules.
3. Supress scancode actions and activate layers and possibly modifiers according to the new keymap. This works in all three cases but is somehow inconsistent with layer switching policy. 
4. Suppress everything but branch on this specific usecase.
5. Some combination of the above.

Ironically even though I started #146 because this exact usecase was broken, I came to an end that I was happy with the option 1. After all, it is consistent across all three cases and makes sense in the bigger picture. And releasing the layer key between keymaps is not that much work after all.

> @kareltucek Wow, you're unusually thorough. Looks like this feature will get tested for good. :)

> So far, only case 1 works. I think it's by far the most common use case.

Actually I am not, I just came across both cases in my configuration by accident.

mhantsch, 2018-10-09:
My two cents on this:

@kareltucek I think case 2 is useful and can be totally consistent. And I strongly believe a held down modifier should still act as a modifier on keymap changes. Rotating through keymaps with something like Fn-+ is useful. Let me explain how I think this could workk.

As soon as the secondary function triggers (either because of timeout or because another key was pressed), the primary cannot activate again until the key is released. So as long as this key is kept held down it now has its secondary function. If the keymap changes while the key is being held, there are some sub-cases:

- a) the key has the same secondary function on the new keymap.
- b) the key has a different secondary function in the new keymap.
- c) the key does not have a secondary function in the new keymap.

In any case, I believe the key - as long as it is held down - can no longer activate a primary function. So, once the keymap switches, it's still a secondary function. Thus:
- in case a: the key still works as the same secondary function,
- in case b: the key now works as the new keymap's secondary function,
- in case c: there is no secondary function on this key, so it does not do anything until released.

kareltucek, 2018-10-09:
@mhantsch Do you mean the "cycling via layer toggled by secondary switcher" case or the "Supress nothing" case? :-)

mhantsch, 2018-10-09:
@kareltucek cycling via layer toggled by secondary switcher

mhantsch, 2018-10-09:
@kareltucek But I think I can be convinced that case 2 "Supress nothing" actually works for case 2 "cycling via layer toggled by secondary switcher" :-D

With this implementation, the a/b/c cases would resolve as:
- in case a: the key still works as the same secondary function,
- in case b: the key still works as the same secondary function,
- in case c: the key still works as the same secondary function,

And that is probably easier to implement. I think it would be logical too: on press, the current keymap decides on the primary and secondary functions. Even if subsequent combined taps on other keys activate other layers or other keymaps, the primary/secondary functions do not change until the held-down key is actually released. Only then on a subsequent press that same key would activate the new primary/secondary based on the now active layer and keymap.

mhantsch, 2018-10-10:
Just updated to firmware 8.5.2. 

Just to confirm: Cycling keymaps works as long as the layer switch is on a primary key. As soon as the cycle hits a keymap where the layer switch is only on the secondary, the cycling stops.

My config:
Right Fn key maps to Fn layer as primary on all keymaps but one.
On the one special keymap, right Fn maps to Fn layer as secondary only, and primary is a regular key.
On each keymap's Fn layer, the ISO key cycles to the next keymap in the sequence.

PR #186 (2018-10-06): Extend the behavior of keys upon layer changes to keymap changes
This closes #167.
closed 2020-04-29 by mondalaci