UltimateHackingKeyboard/firmware #169: Modify shortcut key modifier stick behavior

mondalaci, 2018-09-03:
The modifiers of a shortcut key should stick only as long as as the current layer switcher key is being held. On the base layer, no layer switcher key is held, so the modifiers of shortcuts should not stick.

See https://github.com/UltimateHackingKeyboard/firmware/pull/168#issuecomment-418000863

eltang, 2018-09-03:
How should they behave when a layer is toggled on?

mondalaci, 2018-09-03:
It shouldn't stick if the layer is toggled on. Only when it's held.

luteijn, 2018-09-03:
Since both ways of handeling this will be implemented anyway, please consider making it an option to configure the behavior for those of us that don't want modifiers to stick, or also want them to stick when the layer is toggled on. To avoid impact on the configuration format, perhaps consider some sort of non-serialized configuration section that just has key-value pairs parsed by the firmware (ignoring unknown keys for compatibility), to keep these kind of settings in until the next configuration version update where they can be folded into the main configuration.

mondalaci, 2018-09-03:
We won't introduce key-value pairs because it doesn't fit well into the existing configuration format. Making modifers not stick is possible by implementing the relevant shortcuts as macros. Admittedly, this is a workaround. We might provide a better way of handling this eventually.

dkilmer-mfj, 2018-09-05:
I don't understand the argument for having sticking be the default behavior when holding a layer-switching key (let alone a behavior that requires a workaround to avoid). The argument against it is that it assumes a use-case for layers that isn't universal, acts in a context-dependent way, and has side-effects that aren't obvious to the person setting up a keyboard mapping. All of those arguments are obviated by an option to disable sticking.

My use case is that the Mod layer is for text editing/manipulation. So I have arrows, backspace and so forth mapped there, as well as things like Cmd-C, Cmd-V, Cmd-A, etc. Any time I use a key that has Cmd as a modifier, it carries over to my next press of an arrow key, taking me to the top or bottom of a document. To avoid this behavior I have to map 11 different keys on the Mod layer to macros on the Mac.

I think the option to disable sticking should be a prerequisite to implementing the feature, not something to maybe do later.

mondalaci, 2018-09-05:
@dkilmer-mfj I don't think sticking affects your use case given that modifiers get unstick as soon as another key gets pressed.

kareltucek, 2018-09-05:
@dkilmer-mfj are you on the newest firmware? (8.4.5) This functionality used to be buggy...

dkilmer-mfj, 2018-09-05:
I have never been so happy to have misunderstood something :). The latest version of the agent (1.2.8) is bundled with 8.2.5, which is what I'm using.

dkilmer-mfj, 2018-09-06:
Still happens with 8.4.5. Seems to be Mac only (similar use case on Linux - Ctl-C and Home both mapped to keys on mod layer, pressing those in sequence with mod held down does not go to top of the document). I can create an issue for it.

mondalaci, 2018-09-09:
#170 has been opened by @dkilmer-mfj in the meantime containing a detailed description. Let's not talk about it in this issue and handle it separately from now on.

owlstead, 2018-10-22:
Sorry, but that one is now closed and this one is still open, so what is the status of this bug?

PR #203 (2018-12-26): Adjust the behavior of sticky modifiers
This closes #169 and makes sticky modifiers play nice with secondary role keys. This also removes key suppression, which no longer serves a purpose.
closed 2018-12-26 by mondalaci